vlachoudis/bCNC #586: Generated g-code is always flat

Harvie, 2017-05-13:
Hello,
i've tried several versions of bCNC... namely 0.9.7 and latest git (r911.4761947).
Doesn't matter what i do, i can never get bCNC to generate g-code in Z axis. Eg. when i use plugin to generate bowl it ends up being completely flat in XY plane.

When i try to cut shape from DXF, eg. profile or pocket. It just makes flat g-code. Then i try to apply "cut" on that generated path and nothing really happens. I've set stock thickness, depth increment and everything else. I don't know if it's my fault, but it seems there's just some bug.

Only thing that generates proper code cutting down in Z axis is box generator...

vlachoudis, 2017-05-14:
Can you send us your .bCNC ini, it seems that something is wrong in your settings.

Harvie, 2017-05-14:
Wow! You were right! I've deleted it and it works. so awesome! I've done diff between old and new config and now i figured the problem out.

I had the laser cutter mode on. I've thought it's the problem and i tried to disable it before, but i didn't knew it needs whole bCNC to be restarted before it's actually applied.

closed 2017-05-14 by Harvie

vlachoudis, 2017-05-14:
Good to know!