vlachoudis/bCNC #587: Circles imported from DXF are not really round

Harvie, 2017-05-14:
When i import round object from DXF it's shown as polygon made from straight lines rather than real cirle / arc. I know it probably has to be approximated somehow, but i guess the settings default to something with relatively low resolution/precision. Not sure if that's only displayed like that or it will even be cut square like that, but the idea of running such g-code is bit off puting.

vlachoudis, 2017-05-14:
This is only for drawing and does not affect what it will be machined. You can change the drawing resolution from the ** Tool -> Config -> Plotting Arc accuracy". Splines are different since they are expanded in bCNC and send to grbl as line segments.

closed 2017-05-14 by vlachoudis

Harvie, 2017-05-14:
I've tried plotting arc accuracy 0.01 and 0.001 but it renders still exactly same asi with original setting of 0.1...

vlachoudis, 2017-05-14:
Most probably is not a circle but it is converted to spline. Could you share your dxf file

Harvie, 2017-05-14:
No, it  was the same issue, that configuration was not applied until i restarted bCNC...

onekk, 2017-05-17:
if you are on Linux you have to tweake the .bCNC in your homedir the bCNC.ini in the bCNC directory is only for reference.

vlachoudis, 2017-05-17:
bCNC will overwrite the .bCNC config file at the end when the program is closed. So if you have the program open and at the same time you edit the config file you might lose the changes.

@onekk the bCNC.ini is not for reference. It is the master configuration file. bCNC on load first reads the bCNC.ini and then it overwrites any variable it with the content of .bCNC

onekk, 2017-05-18:
@vlachoudis thanks for the explanation, I've modified my reply to avoid misundestanding.