vlachoudis/bCNC #588: Box generator: add option to make open boxes (like drawers)

Harvie, 2017-05-14:
Please make it possible to generate open boxes with only 5 sides rather than 6. Without finger joints on top side. Eg. something like this:

[image]

vlachoudis, 2017-05-14:
It could be an option for the future. For the moment you can export it as a finger box, save it as DXF, import it in any cad program like librecad, and delete the fingers from the top side. You can even delete the top side and the fingers from inside bCNC but it is a bit more cumbersome.

closed 2017-05-14 by vlachoudis

Harvie, 2017-05-14:
Nice. I didn't realized i can do that. I really like how bCNC does not make difference between g-code and drawing. It makes it really usefull. Like swiss army knife of g-code.