vlachoudis/bCNC #589: Make it possible to set Z jogging step different from XY step

Harvie, 2017-05-14:
There are some portion of machines that have Z axis much shorter than X and Y axes. Eg. my X is 1300mm and Z is only 90mm. For general jogging around it makes sense to have X and Y steps set to 100mm and Z step to 1mm. Can you please add such feature so i don't have to change step lengths for jogging all the time? universal g-code sender has this feature and it's super usefull.

vlachoudis, 2017-05-14:
It's already there, look the issue #49. You need to add a zstep in your ini

closed 2017-05-14 by vlachoudis

Harvie, 2017-05-14:
That's not exactly user friendly...

vlachoudis, 2017-05-14:
Once you enable in the ini you get extra controls

onekk, 2017-05-17:
@vlachoudis is right, but just in case I'm working in a new control panel with some predefined zstep and more z height control. But it has to be reviewed by @vlachoudis, because it slightly modify some internals of bCNC. More to come sooner on this matter.

Harvie, 2017-05-17:
@vlachoudis cool. maybe it can be enabled by default, i guess it's similar for everyone. or there can at least be some gui to enable it.
anyway, i've edited the config and restarted bCNC which resulted in config being overwritten back. so i had to edit it again without bCNC open to make it work.

onekk, 2017-05-17:
The config are not overwritten, but in case when I'm tweaking .bCNC in the user directory I prior make a copy .bCNC-old or .bCNC-YYYYMMAA just in case something go wrong.

Eventually some copy and paste will save you some tedious work, or better use a diff program like meld (if you are a Linux guy) and everything go at his proper place.
Regards