vlachoudis/bCNC #590: Add support for helical cutting / ramp plunging

Harvie, 2017-05-14:
In lots of cases it's much better practice to plunge down along with cutting in X/Y axes, so you prevent mill from going straight down. For some tools it's even the only option... Can you please add this as option for CUT operation?

Eg. when you cutting circle. You don't go down in Z and then do XY circle and then go down in Z again. Instead you can do helix cut down with last layer being the only flat one.

Fosforitofercho, 2017-11-01:
It is a feature of your g-code generator, Use Fusion360, has a LOT!!!! of features like this

Harvie, 2018-02-16:
@Fosforitoferch I hope i will see this feature in bCNC. Fusion360 does not work on Linux. Also i guess this is quite OT.

Harvie, 2018-03-08:
It seems to me, that related code is located in `./CNC.py` file in following methods:
```
def fromPath(
def cutPath(
def cut(
```

Harvie, 2018-06-27:
I am not really python guy and i don't really understand inner workings of bCNC, but i've tried to modify these two functions and get some results. This is image of helicaly cut rectangle. I just selected rectangle from DXF and clicked on CUT button in bCNC:

[image]

Here is the code:

fromPath()

```python
	#----------------------------------------------------------------------
	# create a block from Path
	#----------------------------------------------------------------------
	def fromPath(self, path, block=None, z=None, entry=True, exit=True, zinit=0):
		if block is None:
			if isinstance(path, Path):
				block = Block(path.name)
			else:
				block = Block(path[0].name)

		def addSegment(segment, z=None):
			x,y = segment.B
			if segment.type == Segment.LINE:
				x,y = segment.B
				#block.append("g1 %s %s"%(self.fmt("x",x,7),self.fmt("y",y,7)))
				if z is None: block.append("g1 %s %s"%(self.fmt("x",x,7),self.fmt("y",y,7)))
				else: block.append("g1 %s %s %s"%(self.fmt("x",x,7),self.fmt("y",y,7),self.fmt("z",z,7)))

			elif segment.type in (Segment.CW, Segment.CCW):
				ij = segment.C - segment.A
				if abs(ij[0])<1e-5: ij[0] = 0.
				if abs(ij[1])<1e-5: ij[1] = 0.
				block.append("g%d %s %s %s %s" % \
					(segment.type,
					 self.fmt("x",x,7), self.fmt("y",y,7),
					 self.fmt("i",ij[0],7),self.fmt("j",ij[1],7)))

		if isinstance(path, Path):
			x,y = path[0].A
			if z is None: z = self.cnc["surface"]
			if entry:
				block.append("g0 %s %s"%(self.fmt("x",x,7),self.fmt("y",y,7)))
			#zinit = 0;
			block.append(CNC.zenter(zinit))
			setfeed = True
			prevInside = None
			#print(path.length())
			print("");
			for segment in path:
				#print(segment.length())
				#print(100*segment.length()/path.length())
				if zinit is not None:
					zinit = zinit + (segment.length()/path.length())*z
					print(zinit)
				if prevInside is not segment._inside:
					if segment._inside is None:
						block.append(CNC.zenter(z))
						setfeed = True
					elif segment._inside.z > z:
						block.append(CNC.zexit(segment._inside.z))
						setfeed = True
					prevInside = segment._inside
				addSegment(segment, zinit)
#				x,y = segment.B
#				if segment.type == Segment.LINE:
#					x,y = segment.B
#					block.append("g1 %s %s"%(self.fmt("x",x,7),self.fmt("y",y,7)))
#				elif segment.type in (Segment.CW, Segment.CCW):
#					ij = segment.C - segment.A
#					if abs(ij[0])<1e-5: ij[0] = 0.
#					if abs(ij[1])<1e-5: ij[1] = 0.
#					block.append("g%d %s %s %s %s" % \
#						(segment.type,
#						 self.fmt("x",x,7), self.fmt("y",y,7),
#						 self.fmt("i",ij[0],7),self.fmt("j",ij[1],7)))

				if setfeed:
					block[-1] += " %s"%(self.fmt("f",self.cnc["cutfeed"]))
					setfeed = False
			if exit:
				block.append(CNC.zsafe())
		else:
			for p in path:
				self.fromPath(p, block)
		return block

```

cutPath()

```python
	#----------------------------------------------------------------------
	# Perform a cut on a path an add it to block
	# @param newblock O	block to add the cut paths
	# @param block	I	existing block
	# @param path	I	path to cut
	# @param z	I	starting z surface
	# @param depth	I	ending depth
	# @param stepz	I	stepping in z
	#----------------------------------------------------------------------
	def cutPath(self, newblock, block, path, z, depth, stepz):
		closed = path.isClosed()
		entry  = True
		exit   = False

		# Mark in which tab we are inside
		if block.tabs:
			# Mark everything as outside
			for tab in block.tabs:
				tab.create(CNC.vars["diameter"])
				tab.split(path)

		zinit = 0
		while z > depth:
			z = max(z-stepz, depth)
			if not closed:
				# on open paths always enter exit
				entry = exit = True
			elif abs(z-depth)<1e-7:
				# last pass
				exit = True

			self.fromPath(path, newblock, z, entry, exit, zinit)
			zinit = z
			entry = False
		return newblock
```

Can you please help me to refine this code to make it usable?

Harvie, 2018-06-27:
When i manualy add `z-2` to circle it seems to work... I am not sure why it does not work using `block.append()`. Is there generic way to override Z value of segment? I guess this is the biggest problem.

Harvie, 2018-06-29:
Aha! Finaly got it working!!!

[image]

Harvie, 2018-06-29:
Biggest problem now is that it breaks original CUT functionality. I need to modify the code, so we can switch between helical and not helical cut...

Harvie, 2018-06-29:
Added it to the GUI as an option! I am working on pull request now! :-)

[image]

Harvie, 2018-06-29:
It's out! :-)

Harvie, 2018-06-30:
Just added proper entry and exit paths to avoid crashes.

Harvie, 2018-07-02:
Tabs now work in helical cuting!

Also i got ramp cutting working! It's like helical, but ramps down faster and then continues to cut flat as usual:

[image]

Everything is ready in pull #893

Harvie, 2018-10-29:
I guess this is implemented by now

closed 2018-10-29 by Harvie