vlachoudis/bCNC #591: Enable 3D view to be freely rotated using mouse

Harvie, 2017-05-14:
It can be really usefull to be able to freely rotate 3D view by mouse draging. The predefined ISO1-3 views are kinda doing the job, but it helps to see objects in slight movement to get faster grasp of their shapes.

vlachoudis, 2017-05-15:
I am doing the 3D rendering manually (not using any 3d library) and python is not very fast on that.
I could potentially do it but it wont be interactive.

Harvie, 2017-05-20:
Is there reason for that? I guess there are OpenGL bindings for python, or something like that...

vlachoudis, 2017-05-22:
Yes, bCNC is not using OpenGL, it uses the minimum on external dependencies.

Harvie, 2017-05-28:
I know, but is there sensible reason for that? This "not invented here" approach is not always that wise. Especialy when it not only lacks basic rotating functionality, but you obviously run to performance problems (timeouting renders), etc...

vlachoudis, 2017-05-30:
I would be more than happy if you could help in adding OpenGL as a rendered.

Harvie, 2018-07-08:
Do you know which methods need to be ported to OpenGL?

Harvie, 2018-07-09:
Just accidentaly found this: https://docs.enthought.com/mayavi/mayavi/

Harvie, 2018-08-17:
VisPy seems to have all we need, but i can't figure out how to integrate it to the tkinter window :-(

http://vispy.org https://github.com/vispy/vispy
https://glumpy.github.io/ https://github.com/glumpy/glumpy

Harvie, 2018-08-17:
Seems to me that Tkinter is horrible choice for writing GL applications. We'll have to reinvent some basics to get it working.

Harvie, 2018-08-17:
https://github.com/jonwright/pyopengltk

Harvie, 2018-08-17:
Basicaly only way of having OpenGL in Tkinter is using some 10 years old library called Togl, which is b*tch to install and distribute. Such a PITA.

Harvie, 2018-08-17:
https://stackoverflow.com/questions/8584272/using-pygame-features-in-tkinter
This seems to enable us to embed pygame into tkinter. Probably best we can get with tkinter.

```python
import Tkinter as tk
import os

w, h = 500, 200

# Add a couple widgets. We're going to put pygame in `embed`.
root = tk.Tk()
embed = tk.Frame(root, width=w, height=h)
embed.pack()
text = tk.Button(root, text='Blah.')
text.pack()

# Tell pygame's SDL window which window ID to use    
os.environ['SDL_WINDOWID'] = str(embed.winfo_id())

# The wxPython wiki says you might need the following line on Windows
# (http://wiki.wxpython.org/IntegratingPyGame).
#os.environ['SDL_VIDEODRIVER'] = 'windib'

# Show the window so it's assigned an ID.
root.update()

# Usual pygame initialization
import pygame as pg
pg.display.init()
screen = pg.display.set_mode((w,h))

pos = 0
while 1:
    # Do some pygame stuff
    screen.fill(pg.Color(0,0,0))
    pos = (pos + 1) % screen.get_width()
    pg.draw.circle(screen, pg.Color(255,255,255), (pos,100), 30)

    # Update the pygame display
    pg.display.flip()

    # Update the Tk display
    root.update()
```

Harvie, 2018-08-17:
That basicaly means, you can include any SDL or PyGame application into Tkinter Frame by setting env
SDL_WINDOWID to Frame.winfo_id()

Harvie, 2018-08-17:
Does the Tkinter canvas has to be slow? This seems to be relatively fast:

https://www.youtube.com/watch?v=rkNToAMcIcs

But it does not have antialiasing, which sucks...

Harvie, 2018-08-17:
This does realtime rotation using mouse and does not seem too slow in my opinion:

http://code.activestate.com/recipes/578876-rotatable-tetrahedron/

```python
#!/usr/bin/env python
# coding: UTF-8
#
## @package _12_tet
#
#  Draws a 3D tetrahedron and allows a user to rotate it
#  (mouse left button and wheel).
#
#  The Tetrahedron is represented by a 3 x 4 matrix. 
#  Each column represents a 3D vertex.
#
#  note: a m x n matrix is represented by a list of lines:
#     [[l_1] [l_2] .. [l_m]].
#  m = len(mat), n = len(mat[0]), mat(i,j) = mat[i][j]
#
#  @author Paulo Roma
#  @since 01/05/2014
#  @see http://www.orimosenzon.com/wiki/index.php/Python_samples
#  @see http://mathworld.wolfram.com/RotationMatrix.html

try:
   from tkinter import *     # python 3
except ImportError:
   from Tkinter import *     # python 2
from math import *

def createZeroMat(m,n):
    """Return a matrix (m x n) filled with zeros."""

    ret = [0] * m
    for i in range(m):
        ret[i] = [0] * n
    return ret   

def matMul(mat1, mat2):
    """Return mat1 x mat2 (mat1 multiplied by mat2)."""

    m = len(mat1)
    n = len(mat2[0])
    common = len(mat2)
   
    ret = createZeroMat(m,n)
    if  len(mat1[0]) == len(mat2):
      for i in range(m):
          for j in range(n):
              for k in range(common):
                  ret[i][j] += mat1[i][k] * mat2[k][j]
    return ret

def matTrans(mat):
    """Return mat (n x m) transposed (m x n)."""

    m = len(mat[0])
    n = len(mat)

    ret = createZeroMat(m,n)
    for i in range(m):
        for j in range(n):
            ret[i][j] = mat[j][i]
    return ret

def translate(x,y,dx,dy):
    """Translate vector(x,y) by (dx,dy)."""

    return x+dx, y+dy
   
def drawTet(tet,col):
    """Draw a tetrahedron."""

    w = canvas.winfo_width()/2
    h = canvas.winfo_height()/2
    canvas.delete(ALL) # delete all edges
    nv = len(tet[0])   # number of vertices in tet (4)

    # draw the 6 edges of the tetrahedron
    for p1 in range(nv):
        for p2 in range(p1+1,nv):
            canvas.create_line(translate(tet[0][p1], tet[1][p1], w, h),
                               translate(tet[0][p2], tet[1][p2], w, h), fill = col)

def init():
    """Initialize global variables."""

    global ROT_X, ROT_Y, ROT_Z
    global eps, EPS, tet
    global lastX, lastY, tetColor, bgColor

    tet = matTrans([[0,-100,0],[-100,100,0],[100,100,0],[0,0,200]])
 
    # counter-clockwise rotation about the X axis
    ROT_X = lambda x: matTrans([[1,0,0],           [0,cos(x),-sin(x)], [0,sin(x),cos(x)] ])

    # counter-clockwise rotation about the Y axis
    ROT_Y = lambda y: matTrans([[cos(y),0,sin(y)], [0,1,0],            [-sin(y),0,cos(y)]])

    # counter-clockwise rotation about the Z axis
    ROT_Z = lambda z: matTrans([[cos(z),sin(z),0], [-sin(z),cos(z),0], [0,0,1]])

    eps = lambda d: pi/300 if (d>0) else -pi/300
    EPS = lambda d: d*pi/300

    lastX = 0 
    lastY = 0
    tetColor = 'black'
    bgColor = 'white'

def cbClicked(event):
    """Save current mouse position."""

    global lastX, lastY

    lastX = event.x
    lastY = event.y

def cbMottion(event):
    """Map mouse displacements in Y direction to rotations about X axis,
       and mouse displacements in X direction to rotations about Y axis.""" 

    global tet

    # Y coordinate is upside down
    dx = lastY - event.y 
    tet = matMul(ROT_X(EPS(-dx)),tet)

    dy = lastX - event.x
    tet = matMul(ROT_Y(EPS(dy)),tet)

    drawTet(tet,tetColor)
    cbClicked(event)   

def wheelUp(event):
    """Map mouse wheel up displacements to rotations about Z axis."""

    global tet
    tet = matMul(ROT_Z(EPS(1)),tet)
    drawTet(tet,tetColor)

def wheelDown(event):
    """Map mouse wheel down displacements to rotations about Z axis."""

    global tet
    tet = matMul(ROT_Z(EPS(-1)),tet)
    drawTet(tet,tetColor)

def wheel(event):
    """Map mouse wheel displacements to rotations about Z axis."""

    global tet
    tet = matMul(ROT_Z(EPS(event.delta/120)),tet)
    drawTet(tet,tetColor)

def resize(event):
    """Redraw the tetrahedron, in case of a window change due to user resizing it.""" 

    drawTet(tet,tetColor)
                
def main():
    global canvas
    root = Tk()
    root.title('Tetrahedron')
    root.geometry('+0+0')

    init()

    canvas = Canvas(root, width=400, height=400, background=bgColor)
    canvas.pack(fill=BOTH,expand=YES)               
    canvas.bind("<Button-1>", cbClicked)
    canvas.bind("<B1-Motion>", cbMottion)
    canvas.bind("<Configure>", resize)

    from platform import uname
    os = uname()[0]
    if ( os == "Linux" ):
         canvas.bind('<Button-4>', wheelUp)      # X11
         canvas.bind('<Button-5>', wheelDown)
    elif ( os == "Darwin" ):
         canvas.bind('<MouseWheel>', wheel)      # MacOS
    else: 
         canvas.bind_all('<MouseWheel>', wheel)  # windows

    drawTet(tet,tetColor)

    mainloop()

if __name__=='__main__':
    sys.exit(main())
```

Harvie, 2018-08-17:
Just tried add more 3D lines to that code. It seems to be reasonably fast when working with 10000 3D lines, and i would say that up to 3000 lines it runs almost smoothly.

Harvie, 2018-08-17:
Another solution may be to rewrite bCNC to GTK3 (PyGTK) or wxWidgets (wxPython), which are more up to date than Tkinter and have no troubles with OpenGL.

Harvie, 2018-08-17:
Just tried this and it looks very promissing:

https://github.com/jonwright/pyopengltk

I have OpenGL context in Tkinter window without any hacks!

[image]

Mechscientist, 2025-03-13:
I haven't looked at the image object in tkinter or python's OpenGL interface, but OpenGL can be used to generate an image in an offscreen buffer and then blited to the screen. It might be a possibility to avoid having to rewrite the whole UI for this. There might even be a python package that does this already. Another advantage of doing it this way would be that most modern GUI's allow you to do this image blit so it would be forward compatible should someone attempt to tackle a GUI upgrade. 
Slight tangent to this - the graphics system is pretty rigid in terms of what and how it draws. It seems like the system might be more useable and maintainable if the CNCCanvas object was more of a drawing container/manager and the drawing for the paths, heightmaps, axes, etc. was done by individual classes. This might reduce the complexity of that object, and allow for the addition or substitution of other graphics by plugins.

Harvie, 2025-03-20:
This looks very promissing:

https://pypi.org/project/tkinter-gl/
https://github.com/3-manifolds/tkinter_gl
https://github.com/3-manifolds/tkinter_gl/tree/main/test

Harvie, 2025-03-22:
I've pushed version 0.9.16 to PyPi, i've added tkinter-gl as a dependency. So we can see if bCNC users are able to get that installed. tkinter-gl is [not currently available on ARM](https://github.com/3-manifolds/tkinter_gl/issues/3), but hopefully this will get fixed in near future.

https://pypi.org/project/bCNC/

Harvie, 2025-04-01:
Perhaps if we stick to whatever version(s) WebGL does use, it should be pretty well supported across the platforms right?

Harvie, 2025-04-01:
> I'd be interested in providing assistance.

Thank you. At this point the challenge seems to be to separate the bCNC canvas to self-contained module with sensible API that would be possible to be re-implemented using OpenGL. Like canvas API that would allow multiple geometries (eg. g-code blocks) to be imported in to it and updated. Having callbacks for actions like selecting, copypasting or moving blocks with mouse. Should probably be rather generic (not g-code centric), then it would make sense to make g-code wrapper around it, that would provide abstraction of loading g-code into that. Over all the bCNC canvas API should be decluttered and minimized in a way that would allow for easy reimplementation. If it's gonna be nice, we can even keep it as fallback.

But there are also some ready made OpenGL based visualisation libraries. Maybe we can use some of that without having to implementing everything in house. Might do some research if there is something that we can reause. But that would still need clean bCNC canvas API...

Mechscientist, 2025-04-01:
I agree with you about the API, for a couple of reasons: you can use the current graphics to support systems that don't work with hardware graphics and it might be easier to swap out an OpenGL system for a more modern graphics system like Vulkan at some point.

If you are interested, WebGL has 2 versions: 1.0 and 2.0. Khronos stopped development on it after 2.0 in favor of supporting WebGPU. WebGL 1 is based on OpenGLES 2.0 and WebGL 2 is based on OpenGLES 3.0. OpenGLES is a specification for an interface, and I think that most implementations, other than Linux, are just wrappers over Vulkan or Metal. 

I did a quick search, but couldn't find a python module that provided OpenGL ES contexts, though it was a very quick search.

koneklaudio, 2025-04-16:
Hi, which version of raspberry must bCNC be compatible with? I want to investigate the feasibility of using pyopengltk, which would allow having a 3D canvas with relatively small modifications in the code in CNCCanvas.py (of course, adding all the opengl stuff).
On the other hand, if implementing it on a Pi is not possible, did you consider the option of having a PC version (ideally just modifying CNCCanvas.py)?

Harvie, 2025-04-16:
> Hi, which version of raspberry must bCNC be compatible with?

I think at least RPi 3 and higher would be nice

> I want to investigate the feasibility of using pyopengltk

I think i've already decided to use this https://pypi.org/project/tkinter-gl/

koneklaudio, 2025-04-16:
> I think i've already decided to use this https://pypi.org/project/tkinter-gl/ I've already added it to setup.py to see if there's a backlash.

Ah ok, I didn't know that library.

And do you have some work already done? If not, I could help trying to start to implement it. I am not an expert in python, buy I did some opengl programming with shaders in the past.

Harvie, 2025-04-16:
Please see my comment from 15 days ago. I've described current status there.

We need to refactor the canvas. But yes, your shader skills will probably be useful.

Btw also please take a look at https://github.com/harvie/cnc-simulator  thats another cnc related project heavily centered around shaders, maybe you can suggest some improvements there as well? Shaders are new to me, but im trying to work on that...

Harvie, 2025-06-02:
I've tried tkinter-gl on ARM and it seems to work fine, more details here: https://github.com/3-manifolds/tkinter_gl/issues/3

Harvie, 2025-08-14:
I've tried to query Google Jules AI to port bCNC to OpenGL, it kinda works (displays g-code and allows free rotation using mouse), but LOTS of things are broken. I am not really sure of this is the way, but you can try it here if you want:

https://github.com/Harvie/bCNC/tree/tkinter-gl-migration

please note AI still adds new commits, so if it does not open at all, try older commit...

Harvie, 2025-08-14:
For the rest of you, here's preview video of OpenGL enabled bCNC:

https://github.com/user-attachments/assets/c4792f81-4a27-483d-936f-a72816e71039

Harvie, 2025-08-14:
> Curious how a height map would generate in with OpenGL?

great question, it did omited such feature. however i've asked it to fix the problem, and now it seems there is some viable skeleton, it's not perfect but the data are displayed:

[image]

massiveradiation, 2025-08-14:
Very interesting results!

Harvie, 2025-08-15:
@koneklaudio BTW from overall architecture standpoint, i think the CNCCanvas from original bCNC is not very well designed for efficient GPU acceleration / OpenGL integration. (especially to work with huge g-code files). I think correct solution would be to implement some primitives (eg. lines and groups of lines) which would have internal implementation as shader data. So we don't have to parse g-code again and again unless it was changed. So we can keep g-code blocks converted to that and loaded in GPU memory and work with that as long as possible.

Then we can treat those pre-processed data as objects. We can have callbacks for when someone click on them, or being able just update single g-code block in GPU without having to process all others, etc... Original bCNC have to parse all g-code again every time you change single line, or even just switch the view eg. from XY to ISO1... Which makes working with large files painfully slow.

Have you been thinking about such optimalizations?

Mechscientist, 2025-08-15:
@koneklaudio, this looks really good! But I agree with @Harvie about some restructuring of the canvas if it's desired to keep bCNC going for a while. I'm wondering if you might be able to sift through the AI code and pick up some pointers on text.

Something that comes to my mind in monitoring this discussion is whether implementing the graphics using a scene graph like PyQt (GPL License) would help. I have used it, but a long, long time ago and I'm not familiar with the current implementation details like how it deals with shaders. The advantage is that it gives you the primitive support that @Harvie suggests and it looks like it would provide support for a range of different implementations/versions of OpenGL, though I didn't read through the details of how it handles shaders. For what bCNC is doing in terms of graphics, the shaders may not need to be too sophisticated anyway.

Anyway, food for thought...

koneklaudio, 2025-08-15:
Yes, I though on having classes of primitives. But since I didn't know the original code, I just tried to touch it as less as possible.

I am not sure what would be the best option. An "plain" array of data needs to be created in any case, to be loaded on the GPU. So even if we have an array of primitives, it needs to be converted to the opengl array, to upload it again. So maybe it's faster to manage the plain array directly.

Regarding the change of the view, it does not require updates, as it only requires to update the view matrix.

Please donwload and have a look on the GL branch in my repository. I am not a skilled programmer, so the code might look a bit messy... But please try it, so that you can give your feedback.

The main changes from the original are:

Added opengl folder with shader files
Need pyglm for easier matrix operations
Changed just CNCCanvas.py, and a line in bmain.py

massiveradiation, 2025-08-16:
> Hi,
>
> A couple of months ago I was working on a tkinter_gl version. The idea was
> to modify just the CNCCanvas.py file, keeping the rest as is. It's half
> done, but I see it doable. Today it can rotate and select. There are MANY
> things to do, including height maps (putting text is not straightforward)
> and camera. I believe that the code posted above was generated with
> fixed-function pipeline. I think it's better to use modern opengl (with
> shaders), in order to get a better performance (job done by the gpu).
>
> I did it with GLSL 1.0, to be compatible with PI3, but it's quite
> limiting. It lacks geometry shaders and other advanced stuff that make
> things easier and is more powerful.
>
> I am trying to upload it to github (never did it before...), in case you
> want to take a look. I have no time to advance right now, but I want to
> complete it in September.
>
> This is a recording of what it looks like today:
>
>
> https://github.com/user-attachments/assets/bc56b651-b9d2-4884-8dfe-4247d6bed29c

This is really great progress! I like the idea of keeping the overall
changes to the software minimal.

koneklaudio, 2026-04-15:
> [@koneklaudio](https://github.com/koneklaudio) i've fixed it in your branch: [koneklaudio#4](https://github.com/koneklaudio/bCNC/pull/4)
I don't know, maybe some versions keep compatibility and others don't... Merged!

Regarding the Runtime errors in the picture, it seems that opencv is not compatible with the numpy version installed (camera will not be available). I think python 3.14 is too new for some libraries, so it's better to stick to 3.12-3.13.

Harvie, 2026-04-19:
This is another report i've got from different MacOS user:

```
(venv) user@users-Mac-Pro bCNC-test % python3 -m bCNC
================================================================================
WARNING: bCNC was recently ported to only support 
python3.8 and newer.
Most things seem to work reasonably well.
Please report any issues to: https://github.com/vlachoudis/bCNC/issues
================================================================================
new-config bCNC <configparser.ConfigParser object at 0x10a480c10>
RuntimeError: module compiled against ABI version 0x1000009 but this version of numpy is 0x2000000
RuntimeError: module compiled against ABI version 0x1000009 but this version of numpy is 0x2000000
G2Core loaded
========== OpenGL ==========
OpenGL version: b'2.1 ATI-4.8.101'
GLSL version: b'1.20'
Traceback (most recent call last):
  File "/Users/user/Documents/bCNC-test/venv/lib/python3.11/site-packages/OpenGL/latebind.py", line 43, in call
    return self._finalCall( *args, **named )
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: 'NoneType' object is not callable

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/Users/user/Documents/bCNC-test/bCNC/__main__.py", line 226, in <module>
    main()
  File "/Users/user/Documents/bCNC-test/bCNC/__main__.py", line 153, in main
    application = bmain.Application(className=f"  {Utils.__prg__}  ")
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/user/Documents/bCNC-test/bCNC/bmain.py", line 236, in init
    self.canvasFrame = CNCCanvas.CanvasFrame(frame, self)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/user/Documents/bCNC-test/bCNC/CNCCanvas.py", line 3852, in init
    self.canvas = CNCCanvas(self, app, takefocus=True, background="White")
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/user/Documents/bCNC-test/bCNC/CNCCanvas.py", line 372, in init
    profile_mask = glGetIntegerv(GL_CONTEXT_PROFILE_MASK)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/user/Documents/bCNC-test/venv/lib/python3.11/site-packages/OpenGL/latebind.py", line 47, in call
    return self._finalCall( *args, **named )
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/user/Documents/bCNC-test/venv/lib/python3.11/site-packages/OpenGL/wrapper.py", line 771, in wrapperCall
    raise err
  File "/Users/user/Documents/bCNC-test/venv/lib/python3.11/site-packages/OpenGL/wrapper.py", line 764, in wrapperCall
    result = wrappedOperation(*cArguments)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/user/Documents/bCNC-test/venv/lib/python3.11/site-packages/OpenGL/platform/baseplatform.py", line 487, in call
    return self(*args, **named)
           ^^^^^^^^^^^^^^^^^^^^
  File "/Users/user/Documents/bCNC-test/venv/lib/python3.11/site-packages/OpenGL/error.py", line 230, in glCheckError
    raise self._errorClass(
OpenGL.error.GLError: GLError(
 err = 1280,
 description = b'invalid enumerant',
 baseOperation = glGetIntegerv,
 pyArgs = (
  GL_CONTEXT_PROFILE_MASK,
  <object object at 0x10a0bf200>,
 ),
 cArgs = (
  GL_CONTEXT_PROFILE_MASK,
  array([0], dtype=int32),
 ),
 cArguments = (
  GL_CONTEXT_PROFILE_MASK,
  array([0], dtype=int32),
 )
)
(venv) user@users-Mac-Pro bCNC-test %
```

koneklaudio, 2026-04-27:
Hi, it's actually quite difficult to fix it without having a mac...

Just to summarize the errors:

- numpy version: It looks that python 3.14 comes with numpy 2.x, which is not compatible with the version of some other libraries, that were compiled against numpy 1.x. That can be fixed installing numpy 1.x (pip install "numpy<2")
- cget returning 10c... I believe this happened with the original bCNC, because in the opengl version, winfo_widht() is used, so that should not be happening.
- opengl: I would need to try it by myself, I am not able to fix it otherwise... Also not sure if different versions of MacOS will raise different errors...

I will try to find someone that can lend me a mac...

Harvie, 2026-04-27:
> Also not sure if different versions of MacOS will raise different errors...

I don't think we need to support ancient versions of MacOS, i am fine with merging code that will only run on something reasonably recent...

koneklaudio, 2026-09-08:
> not sure about zooming...need some more time to play with it I have a mac mouse without wheel)
@pauloshendriks , I added the option to zoom with shift+mid button. There is also another way to zoom in/out, already in the original bCNC: z / shift+Z

Apparently, in MacOS + tk < 8.7, middle and right buttons are swapped. I added a check of the running system and tk version, so that buttons are swapped for that config.

koneklaudio, 2026-09-08:
I also implemented a rendered simulation of the stock material for flat and ball nose endmills (3 axis). I am still checking it, but it seems to be working. This option is only available for glsl 1.20 (It should be the version running on all computers, except raspberry pi).

It is based on a heighmap with a resolution of 10000x10000 (The stock is divided in 10000 pieces in both x and y directions). It takes up about 400MB on the GPU. It performs ok in my laptop (gaming), but not sure how it will do in an average computer. Computation of the material removal is done in the GPU.

As of today, the simulation button produces the final shape of the stock, not a realtime animation.

It should be tested with large files, to see if it takes to long to compute the simulation.

https://github.com/user-attachments/assets/13213bd3-01e4-44a5-ab94-6a19992cc2dc

Harvie, 2026-09-09:
> I also implemented a rendered simulation of the stock material for flat and ball nose endmills (3 axis).

banger :-D

> Apparently, in MacOS + tk < 8.7, middle and right buttons are swapped.

Is the general OpenGL support ready to be merged? If so, please prepare some stable OpenGL branch i can review and merge.

@pauloshendriks can you please ack the PR before merging once it's ready?

pauloshendriks, 2026-09-11:
Checked out latest version...findings

Mac mouse has no middle button so panning needs to be done via menu (not really problematic)
Camera connection: USB cam does not always sync properly bCNC picks "at random" one of the connected cams
switching between cameras  possible? Is there a a config possibility for the Cams.
the rest of the tests were as before OK

Question: can I empty the  workspace?

As far as we can see merging the code is OK

Harvie, 2026-09-11:
@pauloshendriks thanks for review. yes, cameras can be configured somewhere in the menu.
@koneklaudio ready when you are...

pauloshendriks, 2026-09-11:
@Harvie : We were looking for that but did not find a selection option

koneklaudio, 2026-09-15:
> Is the general OpenGL support ready to be merged? If so, please prepare some stable OpenGL branch i can review and merge.

@Harvie , I was making minor fixes / improvements during the development of the simulation panel, so now everything is merged in the Master branch. It is working ok on windows and linux. It should be checked again in a raspberry pi to ensure nothing is broken. In any case, the code checks if the app is running on a Pi, and if so, the simulation button is not available.

@pauloshendriks , could you confirm if in the last version (Master branch) the  stock material (when showing the simulation option in the last icon of the canvas) is rendered properly? I tried it in a Hackintosh, and it is not. The sides of the stock sometimes overlap, showing the farthest side on top of the nearest. But the software renderer of Hackintosh is not too reliable, so it should be tested in a real Mac.

I added a couple of options to the 3d view mouse control, in order to ease a bit the use of a 2-button mouse. Now, the rotation center can be changed clicking on a path line either with the middle or right button. And zoom can be also done with shift+middle+drag or shift+right+drag. In any case, i think a 3 button mouse is highly recommended.

pauloshendriks, 2026-09-16:
@koneklaudio 

I will look at it tomorrow

thx

pauloshendriks, 2026-09-16:
[image]

this is what I get, I am not sure what I see here

koneklaudio, 2026-09-16:
It's even worse than I thought... 😅

pauloshendriks, 2026-09-16:
I did not connect to my CNC machine, does that have influence?

pauloshendriks, 2026-09-16:
Can you tell me .......How can I make sure that I run the last edited version?

This is the startup message: ++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++

(venv) paulhendriks@Pauls-AirMac bCNC-master-4 % python -m bCNC                 
================================================================================
WARNING: bCNC was recently ported to only support 
python3.8 and newer.
Most things seem to work reasonably well.
Please report any issues to: https://github.com/vlachoudis/bCNC/issues
================================================================================
new-config bCNC <configparser.ConfigParser object at 0x106b4e3c0>
G2Core loaded
========== OpenGL ==========
OpenGL version: b'2.1 Metal - 90.5'
OpenGL renderer: b'Apple M3'
GLSL version: b'1.20'
Profile mask: ?
Running GLSL 1.20
============================
self.canvas.winfo_id(): 48822321168
Not Loading 6 axis displays
Not Loading 6 axis displays
(venv) paulhendriks@Pauls-AirMac bCNC-master-4 % pwd
/Users/paulhendriks/Downloads/bCNC-master-4
(venv) paulhendriks@Pauls-AirMac bCNC-master-4 % python -m bCNC
================================================================================
WARNING: bCNC was recently ported to only support 
python3.8 and newer.
Most things seem to work reasonably well.
Please report any issues to: https://github.com/vlachoudis/bCNC/issues
================================================================================
new-config bCNC <configparser.ConfigParser object at 0x106e563c0>
G2Core loaded
========== OpenGL ==========
OpenGL version: b'2.1 Metal - 90.5'
OpenGL renderer: b'Apple M3'
GLSL version: b'1.20'
Profile mask: ?
Running GLSL 1.20
============================
self.canvas.winfo_id(): 43233968144
Not Loading 6 axis displays
Not Loading 6 axis displays

Thx Paul

Harvie, 2026-09-16:
honestly i don't care about simulator being broken, should not block merging as long as it does not break any existing feature.
however you might consider separating it to another branch to keep the changes more manageable for review purposes.

BTW stock material size can (and perhaps should) be either autodetected (g-code bounding box code is available + some margin) or loaded from "stock" menu section:

[image]

on the other hand the endmill shape and size should be loaded from metadata introduced by cut operation into g-code comments (or section header). similar to how island/tab metadata are stored there. i guess it might even make sense to support standart g-code toolchange commands. maybe we can just include toolchange command to each cut operation, but leave it commented-out, possibly with extra metadata. so simulator can use it, while keeping compatibility with g-code generated by 3rd party software.

Perhaps we should move discussion on simulator to another github issue.

koneklaudio, 2026-09-17:
> honestly i don't care about simulator being broken, should not block merging as long as it does not break any existing feature. however you might consider separating it to another branch to keep the changes more manageable for review purposes.
> 
> BTW stock material size can (and perhaps should) be either autodetected (g-code bounding box code is available + some margin) or loaded from "stock" menu section:
> 
> <img alt="Image" width="72" height="48" src="https://private-user-images.githubusercontent.com/276504/653031418-e8c6c448-a9b9-4ca3-819a-4074c5780745.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3ODk2MjU4NjAsIm5iZiI6MTc4OTYyNTU2MCwicGF0aCI6Ii8yNzY1MDQvNjUzMDMxNDE4LWU4YzZjNDQ4LWE5YjktNGNhMy04MTlhLTQwNzRjNTc4MDc0NS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjYwOTE3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI2MDkxN1QwNjEyNDBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT01OTg5ZjBiYjlmYTkzNmQxYzlkNWY1Y2FmNGU2Yjk4OTQ3YjMxYmIyMTQxN2MzNjQ4MTNlYTBjZGVjMzNkYTYwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZyZXNwb25zZS1jb250ZW50LXR5cGU9aW1hZ2UlMkZwbmcifQ.j5m8RVuxsIBB5HoDSBUAry5VU-fqkZuYiWHqi_1QkZQ">
> on the other hand the endmill shape and size should be loaded from metadata introduced by cut operation into g-code comments (or section header). similar to how island/tab metadata are stored there. i guess it might even make sense to support standart g-code toolchange commands. maybe we can just include toolchange command to each cut operation, but leave it commented-out, possibly with extra metadata. so simulator can use it, while keeping compatibility with g-code generated by 3rd party software.
> 
> Perhaps we should move discussion on simulator to another github issue.

Since separating the simulation code lines from the fixes/improvements would take time, what I did is to disable the simulation button by default. It will only be shown if the app is started with the --enablesimulation option. (python -m bCNC --enablesimulation)

The simulation opengl shaders initialization is not done until the simulation button is pressed, so starting the app without that option should work exactly as before.

I also thought about the option of linking the simulation tool to the End Mill database as you said, but it is not straightforward. Apparently, that database is a dictionary based on the tool name, but it is not linked to a tool number to be called in gcode with T1, T2 etc. 

Regarding the stock dimensions, in the original code, it does not store the x and y limits, so it should also be completed.

I will create a new issue as you propose, so that we can discuss about it.

Thanks!

koneklaudio, 2026-09-17:
@pauloshendriks , if you clone / download the master branch, it should be the last version. I made a couple of changes. Please try now...
You don't need to be connected to the machine.

As I posted in the last message, now you need to start the app with the --enablesimulation option, in order to show the simulation button in the Canvas toolbar:

python -m bCNC --enablesimulation
or
python bCNC.py --enablesimulation

whichever you typically use.

I hope we won't require too many trials... ;p

Harvie, 2026-09-17:
> The simulation opengl shaders initialization is not done until the simulation button is pressed, so starting the app without that option should work exactly as before.

This sounds perfectly fine. The `--enablesimulation` is not even necessary if simulator will not lead to crash unless the button is clicked. i was more worried about the amount of code i will have to read before merging rather than anything else :-) Sometimes the amount of changes gets quite overwhelming and i don't have that much free time to do the reviews. So i always strongly prefer multiple separate pull requests. As a rule of thumb: Faster i can read and understand the code, the more likely i am to merge it. If i can do it during a lunch break, then i can merge almost immediately. Bonus point if it's a separate plugin file, that i can merge because it's not likely to break anything.

> Since separating the simulation code lines from the fixes/improvements would take time

ok, don't worry about it, this time. OpenGL support is good enough incentive for me :-)

> Apparently, that database is a dictionary based on the tool name, but it is not linked to a tool number to be called in gcode with T1, T2 etc.

Sounds like something we might want to improve in the future. But as i said, we can also encode some metadata into comments as we already do with tabs/islands.

koneklaudio, 2026-10-05:
@Harvie , next step? Shoud I do a Pull Request? If so, to the Master branch?
I guess the classic bCNC should remain somehow available, because some people could have self done code variations that would be broken with the new Canvas.

Harvie, 2026-10-05:
@koneklaudio Yeah, sure. PR would do.

> to the Master branch?

yes, but don't worry, maintainer of receiving repo can change the target branch when needed.

> I guess the classic bCNC should remain somehow available

i will bump the major version and old versions will still be available via their respective git tags.

> because some people could have self done code variations that would be broken with the new Canvas.

they should rebase. we barely have enough capacity to maintain/support the latest version. there is no way of supporting older versions.

Harvie, 2026-10-08:
@koneklaudio side note for the simulator:   how much GPU RAM does the simulation need? Maybe the resolution is just bit too high. Yesterday i was coincidentally looking up something about GPU mem of my raspi4 and the default config was 75 MB. While according to AI, your simulation might need hundreds of MB.

Take a look at this for reference:  https://github.com/harvie/cnc-simulator

[32 more replies]

closed 2026-10-19 by Harvie