vlachoudis/bCNC #592: Enabling serial over ethernet for bcnc

kkugelblitz, 2017-05-18:
This might be of relevance only to a limited audience, but by changing a single line in Sender.py, bcnc can be enabled for serial-over-ethernet connection (i.e. lan/wlan-interfaced grbl by using a tiny protocol server). 

Full support is already contained in pyserial, only the instantiation call [serial.Serial()](https://github.com/vlachoudis/bCNC/blob/master/Sender.py#L536) has to be replaced by serial.serial_for_url() while keeping the parameters identical (I had to increase the timeout parameter, but this might be due to the poor wlan performance at my place). Using local interfaces (comX, /dev/ttyX) works as before as pyserial resolves this internally. 

However, by typing something along the lines of "socket://192.168.1.2:2001" into bcnc's port field (File ribbon), a connection can be established to an ethernet-to-serial server. Personally, I am using [ser2net ](https://linux.die.net/man/8/ser2net) running on an USB-enabled wall-socket mini router (OpenWRT), but a raspberry or any other linux box would do as well.

I tested this with bcnc on windows7-64, but I am not sure about dependencies (e.g. if a certain (minimum) version of pyserial is required) or about possible regressions with mac, linux, 32bit OS.... In principal it should be transparent, but I don't know if its worth taking the risk of changing the master tree, as (supposedly) only few people would be interested?

vlachoudis, 2017-05-19:
@kkugelblitz this is very nice indeed. There were several requested for that. We can check if the address starts with socket to use the serial_for_url().

kkugelblitz, 2017-05-19:
Actually it is not necessary to do any checking/differentiation of the port field input within bcnc; 
just replace the function call within sender.py and everything should be working as before in case you keep your previous input (e.g. "COM1" on windows or "/dev/tty1" on unix).

The function name "serial_for_url()" is a bit misleading: the function is working out-of-the-box with any of COM1, /dev/tty0,  socket://ip:port and several other local or remote protocols. 

From the [api documentation](http://pyserial.readthedocs.io/en/latest/pyserial_api.html#serial.serial_for_url):
"This factory function is useful when an application wants to support both, local ports and remote ports."

vlachoudis, 2017-05-29:
Ok. I've implemented it, could you please check

closed 2017-05-29 by vlachoudis

kkugelblitz, 2017-05-30:
I did some basic testing using my serial-over-wlan installation (run gbrl check function on huge gcode; machine some smaller files) with bcnc running on Windows7 and Ubuntu16.04. 
Everything seems to work fine.

What I could not do is check for regressions i.e. if the "normal setup" with a local serial interface is unaffected, but I suppose you already tried this with your machine?

Thanks.

kkugelblitz, 2017-05-30:
For the sake of completeness: 
I am running a stm32-port of grbl which does not implement the original's "reset-by-DTR line".
So I was not able to check if this works via Ethernet (should, according to pyserial doc).