vlachoudis/bCNC #593: bCNC+Midi2CNC spotted on TV!

CarlosGS, 2017-05-19:
Not an issue - I just wanted to let you know of the (brief) appearance of bCNC + Midi2CNC last night on TV!
Our friend @LuisDiazUgena took his Cyclone PCB Factory to the show and demonstrated how music can be generated with stepper motors :-D

[image]

[image]

You can see the full segment here: <http://www.atresplayer.com/television/programas/hormiguero/temporada-6/capitulo-130-jessica-chastain_2017051800547.html> (about minute 39 into the show)

Luis asked me for the best way to achieve stepper music and I told him about the Midi2CNC plugin. He required no further help :-)
Great job @vlachoudis @Effer @others making CNC so easy!!

LuisDiazUgena, 2017-05-19:
Thanks for sharing that :) it's been nice to bring some open source technology on air!

vlachoudis, 2017-05-19:
Cool! Thanks for sharing. To see the video it requires registering, would it be possible to extract only that portion from the video and upload it on youtube?

Effer, 2017-05-19:
Awesome! Thanks @CarlosGS to point it out! TV in Spain seems lot more interesting than here in Italy :)

CarlosGS, 2017-05-20:
Here you have the segment: https://youtu.be/t4Pr6p_o2Zo

Sorry I forgot of the player restrictions outside of spain :-)

<https://github.com/vlachoudis/bCNC/issues/593#issuecomment-302776880>, or mute
the thread

closed 2017-05-22 by CarlosGS