vlachoudis/bCNC #595: An issue with probing

Avalonnw, 2017-05-20:
I noticed that probing parameters (I tried speed, tool change coords and probing coords) won't apply until software restart. Is that supposed to work like that?

vlachoudis, 2017-05-22:
Nope. They should have an immediate effect. Watch the output in the terminal for any traceback message.

Avalonnw, 2017-05-25:
No output there. I will try to monitor this behavior tonight.

eyal0, 2017-05-25:
This can happen if you change the parameters without pushing the tool change or calibrate button.  For example:

1. Set feed to 1000.
2. Push calibrate. It will work at 1000.
3. Set speed to 10.
4. Push play on gcode that has M6 in it.
5. When M6 is encountered, feed rate will still be 1000.

That has been my experience. It is because the probeUpdate only happens when pushing calibrate or tool change and not from M6 tool change command. Bummer.

eyal0, 2017-05-26:
Try https://github.com/eyal0/bCNC/tree/probe_update_on_change if you like.  It has a fix.

The master branch also has a feature to allow you to have probe with decreasing speed (first probe at speed 1000, then 100, then 10).  Same accuracy but faster tool change.

vlachoudis, 2017-05-29:
@eyal0 thanks for spotting. I will check both patches on my machine

closed 2017-06-07 by vlachoudis

Avalonnw, 2017-06-13:
I tried the new version and had to go back to old one.
The new version does probing, then moves off a bit and then goes to alarm state saying the travel requested exceeds machine limits.
It might be related to my other issue with tool change mid-program as after probing I get the same message. I take it's something to do with GRBL itself (rounding problem?) but have no idea how to rid of this problem. If anyone could help, I would be grateful!

eyal0, 2017-06-19:
@Avalonnw Can you check the terminal output and see what commands are being run?  Maybe you set the probe start or probe distance poorly and the tool is going out of bounds during the probing?

eyal0, 2017-12-08:
@vlachoudis I see that https://github.com/eyal0/bCNC/tree/probe_update_on_change was never merged.  Maybe it's not needed anymore?  I will check using the [steps listed above](https://github.com/vlachoudis/bCNC/issues/595#issuecomment-303936858) and let you know.

eyal0, 2017-12-10:
I just tested on my CNC router and the bug is still there.  I'll resend that PR.

Avalonnw, 2017-12-11:
It looks like there is a bug in GRBL itself where it rounds off steps somehow. You need to switch soft limits off for it to behave properly.

eyal0, 2017-12-12:
@Avalonnw Can you explain?  I'm using grbl and I haven't seen any problems with it.  Maybe send a log from the terminal?

Avalonnw, 2017-12-12:
I'm sorry, it's a bit hard to do a log now, but I can try next time I use my router. It's been a while...
In short, I was having a problem with old style probing if I had to change tools mid-program, as it was trying to move a bit over 0 on Z Axis. Same happens during new probing.
https://github.com/gnea/grbl/issues/252
Removing soft limits helps.