vlachoudis/bCNC #597: Hold Down Tabs and Bridges not manifesting while cutting.

EducatingSavvas, 2017-05-27:
Hi, I am stuck trying to get hold down tabs to work with bCNC. I am trying to do this while importing *.*tap files generated in another software with bridges, or adding tabs to gcode I've opened in bCNC under the tools section. Any suggestions would be welcomed. Thanks.

vlachoudis, 2017-05-29:
Maybe it is just a problem of rendering timeout. Increase the Timeout on the top-right corner of the canvas
to a higher value a check.

EducatingSavvas, 2017-05-30:
Hi - thanks for the suggestion. I think I solved creating tabs in the external software - this was just procedural but I am not sure why the tabs feature didn't work within bCNC. Maybe I would have to tool a job up in bCNC itself? Only problem now is if I import DXF vector files they lines or shapes seem to orientate next to each other and not int heir original positions. Is there a DXF file format preferred by bCNC? Anyway, thanks again.

vlachoudis, 2017-06-07:
Can you send the problematic dxf to check?

closed 2017-06-22 by vlachoudis

kworsfold, 2017-10-28:
I am have same problem..... Created my own .nc file... Edited in bCNC and added Tabs.
Tabs appear in the .nc file as '(Block-tab: 69.1408 99.5994 5 5 -3.75)' and shows as orange boxes on the editor. However when cutting job the tabs do not bridge.
Any further suggestions
[Christmas Hearts.nc.txt](https://github.com/vlachoudis/bCNC/files/1424597/Christmas.Hearts.nc.txt)

(Text file for uploading purposes only)

EducatingSavvas, 2017-10-29:
[image]

The tabs look like they are on the visualisation which mean they should cut. It might be the thickness of your material or the bit is travelling further down the z axis than intended? Does your bit end up cutting into the wasteboard - and if so by how much?