vlachoudis/bCNC #598: How to avoid Z axis crashing? And breaking bits?

asking23, 2017-05-28:
Hi, 

Is there any module or any modification in bCNC which will open dialogue box and confirm" are you sure?  Z axis will go - ve 35mm" before directly hammering into the base of cnc? I have broken many bits while learning got thought when we press start button it should first warn that your z axis will move this much then after confirming it should go down. If anyone has such mods already installed with bCNC please do share with me.

jeffmd, 2017-05-28:
That would be a nice feature.  Currently I just switch my view to XZ and look to see if any tool paths draw below my minimum z limit.

asking23, 2017-05-28:
Can you please help me how to set minimum Z limit? How to view in XZ axis?

vlachoudis, 2017-05-29:
@asking23 a warning could be added for the Z motion if the target is negative coordinates <0.0 but only during the jogging. After pressing the start button, is very tricky to add such a warning.

J-Dunn, 2017-06-07:
If I've understood your problem, you could probably use GRBL dry run mode to check this. It runs without moving anything.

asking23, 2017-06-07:
how to use Dry Run Mode ?

J-Dunn, 2017-06-07:
Check g-code mode : '$C'  since v0.8

kees1948, 2017-08-24:
My biggest problem is that with the 'control values' at 10 (in mm), when the file is loaded and the units switches to inches, one click suddenly becomes 10 inches (255 mm).....

vlachoudis, 2017-08-25:
I am sorry I always think in metric :)
Anyhow, the default values for the 3 predefined steps you can modify them inside the .bCNC file
look the bCNC.ini. They are under the [Control] section as
step1 = 0.1
step2 = 1
step3 = 10
Just copy them in your .bCNC and modify as you wish

kees1948, 2017-08-25:
Yes, but that does not help me here. Consider the situation: start of bCNC, [State] units blank but equivalent to mm. So moving Z goes in x mm steps. So far so good.
But _sometimes_ the units get changed to inch, I did not do that. Now my Z movement is 25.4 times larger :-(. Question: what can cause the units setting in [State] to change?

vlachoudis, 2017-08-28:
The problem is that bCNC is not aware continuously of the state of grbl. So if something changed the units, bCNC might be unaware the moment you start jogging.

chaotix-, 2017-10-08:
Using GRBL 1.1 new jog commands this could be solved.

https://github.com/gnea/grbl/wiki/Grbl-v1.1-Jogging

`$J=G91 G21 X1.0` will always move to +1.0mm and G21 (or G20 if you really wanted to :-)) would only be active for this jog move so it would not change the state of the machine.

Problem is that this is not compatible with GRBL < 1.1.

vlachoudis, 2017-10-20:
@chaotix- thats a good idea.

Harvie, 2018-10-29:
https://www.youtube.com/watch?v=Q-Q1o16Ux-I