vlachoudis/bCNC #599: Negative autolevel value fails silently

dakahler, 2017-05-28:
It is possible to set an X or Y max value to a negative number, so autoleveling can happen with an origin that's at the positive-most side of the target area. This appears to work just fine - except bCNC just silently fails to record the result of each probe point.

The issue is in probe.add, ~line 298 in CNC.py. The early out tests here assume a positive value, so it silently fails to add the probe point to the list.

vlachoudis, 2017-05-29:
I don't see any restriction in using negative values in xmin, xmax from the routine you mention
```
		i = round((x-self.xmin) / self._xstep)
		if i<0.0 or i>self.xn: return
```
you have to be sure that the self.xmin is smaller than self.xmax

dakahler, 2017-06-03:
I think that was probably the issue. It would be nice if it threw an error one way or another letting me know I'm an idiot, instead of failing silently.

J-Dunn, 2017-06-07:
This seems similar to the issue where GRBL error 20 gets ignored and causes the progess to get stuck at 97% . Would some improved error trapping be a good idea?

vlachoudis, 2017-06-07:
@dakahler I 've just added a pop up a warning message.
@J-Dunn error trapping would require bCNC to know what every version of each controller supports and how it behaves. This is dangerous and risky to make errors. The easiest approach is to enable the gcode checking in grbl, and run your program to see if grbl gives any error code before running.

closed 2017-06-07 by vlachoudis

J-Dunn, 2017-06-07:
I think you are misunderstanding what I mean by error trapping. I am not suggesting it knows what all controllers do or expect, simply that if some error condition occurs in bCNC which causes a dysfunction it should be trapped.  In the error 20 case it does not need to know the details of GRBL, but if something unexpected happens from the context of bCNC, it would be preferable to trap and deal with it in some way rather than leaving the cursor spinning etc. 

Whether GRBL is significant enough to merit making use of the new error return is secondary to preventing bCNC getting into a confused state.

In this issue, the error was internal to bCNC and was trapped but ignored not propagated and it  failed silently. Since the primary result of creating auto-leveling file was not achieved that was a bug. Your pop up solution seems a good fix. Thanks.