vlachoudis/bCNC #602: Can't restart after Pause and illegal character

woodworkerbob, 2017-06-10:
Running bCNC 0.9.9, and Shapeoko grbl 0.9g

This shows the start up:
[StartbCNC.txt](https://github.com/vlachoudis/bCNC/files/1066067/StartbCNC.txt)

Once the Shapeoko is moving I click on the Pause button.  After the GUI buttons: Start is grey; Pause is black; Stop is black; Control Panel is grey.  Clicking on Pause has no effect.

Terminal pad before Pause;
[PausebCNC.txt](https://github.com/vlachoudis/bCNC/files/1066069/PausebCNC.txt)

Since Pause does nothing, I click on Stop and get the following:
[StopbCNC.txt](https://github.com/vlachoudis/bCNC/files/1066070/StopbCNC.txt)

Also note the illegal character.

Can I provide more information to help?

Bob

brianhaupt, 2018-01-04:
Has this bug had any action?  I am having the same problem with bCNC 0.9.11 and Grbl 1.1e

I don't see it sending any commands for the pause and I am not really expecting any action as I don't use Grbl for spindle control, but it would be really nice to be able to resume.  The Stop button also has no effect after trying to resume.  Currently I have to restart bCNC, modify my cut file in the editor, realign the workpiece with the tool head, and then start the modified job.

I would be happy to look into this although my Python experience is very limited.  Please let me know if I can provide any more information to assist in fixing this issue...

Thanks,
Brian

brianhaupt, 2018-01-04:
I apologize for any confusion, but earlier this evening I believe I had the problem stated above (as I posted earlier).  After trying to go in and fix it myself, the pause/resume function started working but I had backed all of my changes out.  Thinking it might be a distribution problem, I extracted the original distribution to a clean folder and found that it appears to work there also.  I don't really understand when the python code is compiled.  It seems to recompile anytime I make changes and I see the distribution doesn't include the compiled python files, so I am still trying to figure out what is going on.

vlachoudis, 2018-01-17:
When hitting "Stop" bCNC is forced to send a software reset to grbl, in order to cleanup any pending command in the buffer. Before the reset it records the current setup ($G) and after the reset It waits for some time and then starts sending the $G to restore the state. I believe that the problem happens exactly there, when it tries to restore the settings while maybe arduino did not finish the reset.

Harvie, 2018-08-13:
please try latest git version

closed 2018-10-29 by Harvie