vlachoudis/bCNC #610: Freezing during streaming with stacktrace

twforeman, 2017-06-27:
Running on Windows 10
bCNC v0.9.9
Python version 2.7.13

This is a new laptop for me. bCNC was running fine yesterday then last night Windows did it's update dance and now bCNC starts streaming the file and gets about 14% done and then freezes with a stacktrace in the python window.

```
Exception in thread Thread-2:
Traceback (most recent call last):
  File "C:\Python27\lib\threading.py", line 801, in __bootstrap_inner
    self.run()
  File "C:\Python27\lib\threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "C:\Users\timf\bCNC-0.9.9\Sender.py", line 986, in serialIO
    if self.serial.inWaiting() or tosend is None:
  File "build\bdist.win-amd64\egg\serial\serialutil.py", line 588, in inWaiting
    return self.in_waiting
  File "build\bdist.win-amd64\egg\serial\serialwin32.py", line 257, in in_waiting
    raise SerialException("ClearCommError failed ({!r})".format(ctypes.WinError()))
SerialException: ClearCommError failed (WindowsError(995, 'The I/O operation has been aborted because of either a thread exit or an application request.'))
```
The time counter at the bottom of the window is still running. I suppose it's possible that my Grbl board is freezing up, since the spindle is stopping, but this would be new behaviour.

I'll try reinstalling python just to see if that makes a difference.

twforeman, 2017-06-27:
Just as an update, I have re-installed python and pyserial, and rebooted the laptop. Still freezing. It's not always at the same place, this time it made it 37%.

I'm going to contact Carbide3d since I have one of the original first run controllers.

vlachoudis, 2017-06-27:
@twforeman it seems a hardware or driver error in the communication

electrokean, 2017-06-27:
This sounds a lot like electrical noise causing a USB reset/disconnect.  I suggest you try a different USB cable.  Preferably use one with a ferrite "lump" on it, or at least one that isn't too long (bigger antenna) or too thin (too little copper/shielding).

twforeman, 2017-06-27:
Yeah, it feels like noise to me also. The annoying thing is that nothing has changed. It worked great on Sunday (as in ran for hours with no issues), on Monday it started freezing. The only change was Windows updates.

I'm trying to get a newer board from Carbide3d. The newer ones are less resistant to noise. I'll get a better USB cable too.

electrokean, 2017-06-27:
You mentioned a new laptop, so I thought some of your USB "hardware" had changed.

twforeman, 2017-06-27:
The laptop is new, but I bought it on Sunday and ran the mill for hours with it with no issues.

It's possible the laptop is the problem, I have a different one I can swap in to test with.

twforeman, 2017-06-30:
Received a new CarbideMotion board yesterday. Installed it and successfully ran for an hour and a half.

Looks like it was hardware after all. Thanks!

closed 2017-06-30 by twforeman