vlachoudis/bCNC #611: "no __call__method" when importing certain .dxf-files.

fk47, 2017-06-29:
Hi all,

I'm trying to load a .dxf-file from draftsight but with no success. If I run the file in another CAD-program and save it as a .dxf it however works. This is the traceback:

Traceback:
Traceback (most recent call last):
  File "C:\path\to\bcnc\bCNC-0.9.9\Utils.py", line 397, in __call__
    return self.func(*args)
  File "C:\path\to\bcnc\bCNC-0.9.9\bCNC.py", line 1890, in loadDialog
    if filename: self.load(filename)
  File "C:\path\to\bcnc\bCNC-0.9.9\bCNC.py", line 1951, in load
    Sender.load(self,filename)
  File "C:\path\to\bcnc\bCNC-0.9.9\Sender.py", line 488, in load
    self.gcode.importDXF(filename)
  File "C:\path\to\bcnc\bCNC-0.9.9\CNC.py", line 2363, in importDXF
    dxf.readFile()
  File "C:\path\to\bcnc\bCNC-0.9.9\lib\dxf.py", line 1211, in readFile
    while self.readSection() is not None: pass
  File "C:\path\to\bcnc\bCNC-0.9.9\lib\dxf.py", line 1193, in readSection
    self.readEntities()
  File "C:\path\to\bcnc\bCNC-0.9.9\lib\dxf.py", line 1118, in readEntities
    entity.read(self)
  File "C:\path\to\bcnc\bCNC-0.9.9\lib\dxf.py", line 623, in read
    self._readVertex(dxf)
  File "C:\path\to\bcnc\bCNC-0.9.9\lib\dxf.py", line 599, in _readVertex
    tag,value = dxf()
AttributeError: DXF instance has no __call__ method

Any clue of a solution? Thankful for any help.

vlachoudis, 2017-06-29:
There was a typo/error in dxf.py module. I've corrected it. Can you try now if it works?

fk47, 2017-06-29:
Wow, it does indeed work now with no errors! Thank you for the quick response.

vlachoudis, 2017-06-29:
Perfect, thanks for the bug report.

closed 2017-06-29 by vlachoudis