vlachoudis/bCNC #612: GCode error handling

ericsgear, 2017-07-01:
I'm using bCNC 0.9.9 on a RPi 1b, sending to Grbl 0.9j on an Arduino Uno R3.

If I try GCode that I know works on another machine (Mach something-I-think on LinuxCNC) and the file uses gcode commands that Grbl doesn't support, bCNC seems to pick a random point in the gcode file to skip to and starts or restarts from there.  Sort of.  It has issues because of unset feed/speeds, so it freaks out again.

Another issue that comes up because of this is that the Stop button doesn't stop sending gcodes to the Arduino.  I can hit Stop as many times as I want and it'll maybe stop for 1/2 sec before it continues on.  I have to hit the Pause button, then Reset.  _This is an unsafe fail state._  If I wasn't cutting air while testing a machine I recently built, it could be catastrophic.

bCNC needs a graceful way to fail.  Also, it'd be nice if it would tell the user that their gcode is incompatible with the the GCode parser it's being sent to.  It doesn't even need to suggest how to fix it, just pointing out the line failure would be great.  The "random re/start location" makes it impossible to know what's going on.  I didn't find this out for 100% sure until I tried a gcode file that worked.

I don't know if the Stop button is directly related to the gcode failure handling, but it's the only time I've seen it happen.  Since I can reliably recreate the Stop button issue with the gcode failure mode issue, and I haven't gotten it with a working gcode file, I'm adding it to this issue.

Also seemingly related: after a gcode failure, trying to set the work position (WPOS button) sets the position incorrectly by what feels like an exponentially wrong position.  When trying to position the work position by a small amount, the actual amount changes more drastically wrong each time, leading to the work piece not even being on the virtual bed after just a few tries.

If you need to split these up to 3 separate issues, go ahead.  As a programmer myself, I sometimes like to see a progression like this, so I can fix things in the appropriate order.  I find it annoying to fix one issue, only to find that it makes tracking down another issue nearly impossible.

As a programmer that works in C# for web, Java for Android, and C/C++ for Arduino, let me know if there's something I can check out for you or if you need me to try some work arounds.

Thank you!

Eric Ingamells

vlachoudis, 2017-07-03:
Usually erratic/random behavior is due to hardware problem most probably noise in the line.
If there is a problem in the gcode that grbl doesn't accepts, bCNC should stop always at the same point with the same error message.

bCNC default behavior is to stop any code submission as long an error message is returned from grbl. The only freedom that the user has is to select some g-codes to be skipped and not be send at all to grbl. Also when an error appears in the status, you can click the status button to tell you which code failed and a short explanation of the error code. As well in the terminal it displays the problematic line.

I believe the fact that the WPOS is changing seems that something is bad in the communication. bCNC simply reports the output of a '?' in the MPos,WPos

Last I would strongly suggest upgrading to the latest Grbl 1.1
Can you provide an output of the grbl settings.

ericsgear, 2017-07-04:
Thank you for getting back to me so soon.  Since I don't seem to have described my situation well enough, I made a video with what's going on.

https://youtu.be/8jGnQIX87rs

I've followed your suggestion of upgrading Grbl to 1.1f, without any change.  The video includes the settings, but I'm including them here for simplicity.

$0=6 (step pulse, usec)
$1=25 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=0 (dir port invert mask:00000000)
$4=0 (step enable invert, bool)
$5=1 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$11=0.010 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=1 (soft limits, bool)
$21=1 (hard limits, bool)
$22=1 (homing cycle, bool)
$23=0 (homing dir invert mask:00000000)
$24=100.000 (homing feed, mm/min)
$25=1000.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$100=44.200 (x, step/mm)
$101=44.200 (y, step/mm)
$102=1533.237 (z, step/mm)
$110=35000.000 (x max rate, mm/min)
$111=35000.000 (y max rate, mm/min)
$112=900.000 (z max rate, mm/min)
$120=800.000 (x accel, mm/sec^2)
$121=800.000 (y accel, mm/sec^2)
$122=500.000 (z accel, mm/sec^2)
$130=2463.800 (x max travel, mm)
$131=1244.600 (y max travel, mm)
$132=117.800 (z max travel, mm)

I didn't remember to record the WPOS issue, so I'll just say that's a non-issue for now.

Let me know if there's anything else I can do to help you.

Eric Ingamells

vlachoudis, 2017-07-05:
This is a bug normally after the error should stop completely. Could you send me the gcode to check.

The error 33 means that an arc g2 g3 has a mismatch in the end point and the radius/center provided.
For the missing rendering increase the timeout on the top right corner of the canvas

ericsgear, 2017-07-06:
[original.TXT](https://github.com/vlachoudis/bCNC/files/1126364/original.TXT)

I have so many copies of this file that I'm not sure if it's the one I'm showing in the video, but it should give you the errors I'm seeing.  All of the files I've generated, prior to yesterday, were giving me the errors, so it's a safe bet for me.

Just yesterday, I found a new config settings file for my CAM software which seems to be working for me.  I'm no longer seeing these issues in the new GCode I'm generating, but I'm still willing to help you with testing, etc.

I have another model that uses the same incorrect settings that we can use to do 2nd round testing.  I'm not including that here, since I want to keep that private until I can get some tests recorded and on YouTube first.  Hopefully I can get that done soon, now.

Eric Ingamells

ericsgear, 2017-07-06:
BTW, I was switching between G91.1 and G90.1, so try both to see what's going on.  Since grbl only uses G91.1, I think I was trying to force it to work, but it may only render correctly with G90.1, which I think is what's causing the errors.

Eric Ingamells

vlachoudis, 2017-07-07:
I think I found the problem(s). It was two things
1. After an error I was trying to reset the controller to empty the buffer. However if the last motion was not properly finished grbl was raising an alarm, and the program was getting in  a crazy state and continue to submit commands.
2. the runEnded() call was added some time ago in the error recovery procedure, however it was called before the controller was reset.
Could you check that it works ok now.

ericsgear, 2017-07-08:
I will try to try it out Monday.  Since I know my machine is working correctly, and now that I have a reliable way to make good gcode, I'm doing testing of my own and doing projects.

I also just broke off a pin on my spindle speed control and I need to add in some more noise filtering on my end stops.  Hopefully I can get those addressed tonight yet, then finish up what I'm doing on the CNC.

So yeah, probably Monday.  Sorry for the delay.

Eric Ingamells

ericsgear, 2017-07-10:
As a Linux noob, I had trouble getting the latest version to run, but I eventually got there.

I ran a known bad and a known good gcode file through with good results.

The known bad file stopped on the bad gcode.  When the error happened, I was able to get a human readable error info from the status button in the top left corner.

The known good file ran for 1 min before I manually stopped it.  It would normally fail within 5 sec if it wasn't going to run due to bad gcode.

I had a little trouble with the WPos button creating an unexpected margin around the known good file, after running the known bad file, but after clicking New and reloading the known good file, the margin went away.  I don't know if I can recreate this.

All in all, I'd call the original/main issue fixed.  I've only tested it once with 1 good and 1 bad files, so you may want to do more testing.  I know that my bad files would cause the reported issue every time, so the fact that it didn't do it once is good enough for me.  I may have more time next week to do more testing for you, but otherwise I call this resolved, for me anyway.

I have to get back to getting ready for a show this weekend, otherwise I'd do more testing.

This puts to rest a problem I've been having for 6 months, preventing me from moving forward with my CNC machine, so **thank you very, very much**!  I wish I could have figured out what was happening before now, so I could have reported it to you in January.

Thank you again and thank you for getting this resolved so quickly!

Eric Ingamells
Eric's Gear
EricsGear.com

vlachoudis, 2017-07-11:
Perfect, then I close the issue, and if you discover the other problem, please open a new one

closed 2017-07-11 by vlachoudis