vlachoudis/bCNC #613: Editing the custom script buttons

jamesarm97, 2017-07-04:
Once you configure a quick script button (the ones at the top left), how are you supposed to edit them to make corrections (besides edit the ini)? Seems like a right click option to edit could be enabled. Once a button is configured you can't edit again from what I can tell. Related question, some of the button names if long get chopped off, is there a way to increase the top tool bar height? A drag line or in config?

vlachoudis, 2017-07-05:
With Right-click you can edit them.
The height of the tool bar should be configured automatically by tk, depending on the fonts and the text.
Can you post a screen shot of the problem

closed 2017-08-15 by vlachoudis

jamesarm97, 2017-08-15:
Running on a Mac. Here is a screenshot (of the buttons now showing all info). I also tried right-clicking again (that is why I originally opened the issue, I tried every combination). Right-clicking does nothing. Left clicking executes button and I tried all other combinations I could think of. Is there any debug or info I can look at?
[image]

jamesarm97, 2017-08-15:
I can live with them looking odd (I was actually going to rename them and hence opening this issue). I can probably just edit the ini file but was hoping for an easy way in the future just like when you initially edit the button if not configured.

vlachoudis, 2017-08-15:
Indeed in Mac it look odd. I let free the resizing, so the system is deciding and I've added also the 
Control + Left-Mouse to open the edit dialog as the Right-Mouse click.
Tell me if it works

vlachoudis, 2017-08-15:
I would need to get access to a Mac to debug it there. In the mean time you can always change the settings in the .bCNC file.
Which python/tkinter version you are using in Mac?

jamesarm97, 2017-08-15:
Ok. Im making sure everything is updated. Was running Python 2.7.13. I deleted one of the buttons and Clicking a blank button brings up the editor but CTRL-Clicking the same blank button hangs just like a edited button so something about the click handler calling self.edit must be different.

jamesarm97, 2017-08-15:
I fixed it. Python was using Apple's tcl-tk so I compiled python with a downloaded version. Works without locking up now. Thanks for fixing the ctrl button option.

vlachoudis, 2017-08-15:
Great, maybe you can document your findings in the wiki page on the Mac section.
Did the graphics improve?

jamesarm97, 2017-08-15:
The buttons and everything seem to look the same but it does seem snappier. Launches instantly.

hb020, 2017-08-16:
The Tk versions will remain to be a cause of problems. 

My findings: do NOT use OSX provided python nor tk. Those versions are broken in many ways. One of the options is to use Brew, but with the latest brew you will have to either launch with 'python2', or put the path to brew stuff before the OS supplied path. The published and preferred method of making brew python the main python, namely "brew link python", no longer works as it should, due to an explicit choice by the brew maintainers. (see https://github.com/Homebrew/homebrew-core/issues/15746 if people want to read)

Mac bug on the latest master pull:
when using Ctrl-Click on any of the 3,4,5 buttons, closing the popup by clicking the red 'X' top left of the popup makes it appear once again. That popup can then be closed as usual. This does not happen on Ubuntu.

Minor thing: looks are a bit different:
[image]
[image]

Mac via brew: tcl-tk = 8.6.7
Ubuntu 16.04: tk = 8.6.0+9 and tk8.6 = 8.6.5-1

vlachoudis, 2017-08-16:
Thanks @hb020 could you please edit the https://github.com/vlachoudis/bCNC/wiki/OS the Mac section and add your instructions there.

hb020, 2017-08-16:
done.