vlachoudis/bCNC #614: autoleveling: how to reset the overall height of a probed matrix with a new calibration?

MoonCactus, 2017-07-05:
TL/DR: how to safely re-use the probing matrix when all Z information was lost, including TLO and calibration offset?

**Context**: I milled a PCB with preliminary autoleveling, which I saved in a .probe file, moved to X0 Y0, and I then switched everything and went to bed. A few days later I wanted to resume the work (drilling the PCB). So I loaded the probe file, calibrated my tool and went ahead. Now for some reason my Z was probably changed manually in between, so the former tool offset was lost or wrong (and I am still a newbie).

**Problem**: My problem is now that fail to achieve a meaningful Z=0, even after probing or calibrating a new tool. I I even tried to set the TLO and/or Calibrations values manually, but whatever, bcnc it will always "dig" further in the PCB surface by the value that is shown on the (0,0) probing matrix!

**Question**: Is there a way out beside editing the probing matrix so its value at X0 Y0 is zero? Obvisouly the program could compute the real offset knowing that the X and Y were kept in between (or they could be found out again with an edge finder).

I think it boils down to being able to probe again the the Z origin ... of the probed matrix! Then it would and offset all other values accordingly. But it looks like probing again is clearly not enough, though what is really important when autoleveling are the relative Z offsets given one known Z height and not overall value. The origin is probably better set by probing, or at least it would be cool if I could ;)

In the other case, what kind of workaround can I get? I was thinking about arbitrarily shifting the origin after probing, but this is quite risky given that there are many tool changes in the gcode file, and I am not confident enough to asses the side effects...

vlachoudis, 2017-07-05:
This is exactly the purpose of the "Zero" button in the autolevel group of the ribbon.
You can probe any location for the Z, then without moving the gantry from the X/Y location you click the "Zero" button and it will re-work all the autolevel matrix using this X/Y point as reference for zero.

MoonCactus, 2017-07-05:
Oops! :/ Worst of all I think I did read it in the doc... but I did not understand fully. I tried it too quickly but probably did not understand enough what I was doing (since Z moves back higher after a probe I felt like it would have been a bad idea!)

Sorry and thumbs up again for your reactivity!

closed 2017-07-05 by MoonCactus