vlachoudis/bCNC #615: Runing Gcode problem & plasma or laser cuting

Nossolov, 2017-07-06:
Use bCNC last version with MacOS and grbl0.9 on Mega & Ramps1.4
When i click on the start button to run a G code, sometime it woks but most not. I have the orange line in the bottom of the screen, time goes on, but nothings append. I have to click on the pause button and it goes well. Strange ... or not ?
When you use a DXF file, every contour give a bloc and the Header define in tools/config is add in the G code. Nice ! When you cut with plasma or laser, you have to start plasma at the begining of each contour and wait for some seconds, stop at the end and start again for the next. Is it possible to define  and add automatically a kind of header for every bloc to do this ?
Thanks for creating bCNC ! A great job.

Nossolov, 2017-07-06:
Sorry for laser cuting : i just find the Tools Machine option.  Take a look at this tomorrow

Nossolov, 2017-07-07:
No Tools/Machine option in my MacOS Version, just a box to check in Tools/config.
Dogmany post a clever solution in august 2015. I do that and it's fine for me. May be, it can be done in automatic every time you check "laser" opion ?

vlachoudis, 2017-07-07:
Look at "Tools -> Config". Tick the  "lasercutter" option

Nossolov, 2017-07-10:
Ok, great !
Don't understand why it doesn't work the first time i try. May be, i have not restart bCNC ?
Thank's a lot.

vlachoudis, 2017-07-10:
Its a good habit to restart if you change something in the configuration.

closed 2017-08-03 by vlachoudis