vlachoudis/bCNC #616: Go to study!

ser007, 2017-07-07:
Hi. bCNC is a great project. Thanks for the work!

I'm looking forward to studying it so I can learn how it works and one day I can collaborate.

I would like to know the best development tool (IDE) that is compatible with Python Tinker where I can handle the interface, add more buttons, etc.

Thank you

vlachoudis, 2017-07-08:
You are welcome. I didn't find any developement UI for tkinter. I use vim

closed 2017-07-08 by vlachoudis

Effer, 2017-07-09:
Hi @ser007 , 
just a couple of thoughts:
- The codebase is huge, and I mean HUGE. @vlachoudis wrote thousands of code lines and bCNC do a lots of things. So it's not easy to get the whole picture.
- Doc is poor but the code is well written (and quite self documented). You can learn a lot from a project like this.
- Believe it or not everything in the GUI is handwritten, so there's no tool to visually adjust icons,buttons,layout. Here the code rules.
- As IDE I've used PyScripter, PyCharm and then switch to VS2015 with python plugin (Python Development Tools). All works well, supports breakpoints and watch variables out of the box. For some changes and plugins I've simply use Notepad++.  With python you can do.
- I've found the [plugins ](https://github.com/vlachoudis/bCNC/wiki/How-to-create-a-plugin)sandbox an easy point from where to start. There you can create some specific functionality without wondering too much about the rest.

Hope this could help.

ser007, 2017-07-09:
Hello guys,

@Effer, I'll check the tools you're suggesting.

@vlachoudis, @Effer thank you