vlachoudis/bCNC #617: How do I use the flatten tool?

rhfinch, 2017-07-10:
I have a fealing I'm missing something, but I can't figure out how to use tthe flatten and other tools.

I Enter stock details, material, and end mill the click flatten. I fill in in the flatten panel, then what do I do  to make it happen? Some documents refer to an Execute button but I don't see one. I tried cut and enter more data but nothing happens. What am I missing?

Any advice would be greatly appreciated.

mandrav, 2017-07-11:
Just above the tool settings, there is the title of the tool, "Flatten" in this case, with a gear icon next to it. The title is actually a button: just press it to execute the plugin...

rhfinch, 2017-07-11:
Thank you so much mandrav. It was driving me crazy!

closed 2017-07-11 by vlachoudis