vlachoudis/bCNC #619: Trouble importing files to bcnc on Raspberry Pi 3

suiterec, 2017-07-12:
Hi, I am trying to open .dxf files (exported from Solid Edge) into bcnc on my raspberry pi 3. When I try to open anything, nothing happens. There is loading text in the bottom left of the window with the correct file path, but no matter how long I leave it, nothing ever opens. What could be my problem? Thanks in advance!

vlachoudis, 2017-07-12:
Could you upload one of those dxf files to test.

suiterec, 2017-07-12:
.dxf flies are not supported by the file uploader on github so I have attached it here.

suiterec, 2017-07-12:
Sure. Here is one I have had trouble with. 

[Profile1.zip](https://github.com/vlachoudis/bCNC/files/1143212/Profile1.zip)

vlachoudis, 2017-08-02:
When I load your dxf I get the following image
[image]
which is consistent with what I see in librecad

closed 2017-08-12 by vlachoudis