vlachoudis/bCNC #620: Macro Shortcut Keys

gjcollie, 2017-07-20:
I have tried searching but I can't seem to find any info on this.  Is there a simple way to assign a keyboard shortcut to one of the 6 programable macros/scripts?  I use these to initialize custom workspaces/probe etc.  I'm currently using a PS3 controller as a pendant for my SO3-XXL, I've mapped keyboard commands to the buttons to control jogging homing etc, but I'd like to be able to execute some of the macros using the buttons/keyboard.

Thanks!

vlachoudis, 2017-08-02:
Not for the moment but it is a good idea to add them.

Guenni75, 2018-01-13:
How do you map the Keyboard commands to the PS3 Controller?

einencool, 2018-04-30:
Hello vlachoudis,

can you tell us, if it could be possible to add this feature to your program?
This would be very great and helpful.

Thanks in advance.
Greets Chris

massiveradiation, 2020-04-19:
👍
I could really use this feature too!
Thanks!