vlachoudis/bCNC #622: bCNC crashes with ValueError when running gcode file.

rhfinch, 2017-07-25:
bCNC staops sending gcode to grbl after about 15 minutes. Grbl is still active and connecting a terminal program and entering ? return status if idle.

At the time of failure the Python screen displays the following:

Exception in thread Thread-5:
Traceback (most recent call last):
  File "C:\Python27\lib\threading.py", line 801, in __b
    self.run()
  File "C:\Python27\lib\threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "C:\bCNC-master\Sender.py", line 1007, in serial
    CNC.vars["my"] = float(word[2])
ValueError: invalid literal for float(): -121.04-0.500

Failure is not always in the same place in the file.The same gcode script runs with UGS for hours without failure.

here is the gcode.

[bcnc test.txt](https://github.com/vlachoudis/bCNC/files/1174476/bcnc.test.txt)

electrokean, 2017-07-26:
Looks like your serial communication is losing characters.  Does your Arduino have a CH340G USB to serial chip by any chance?  This type of issue has been seen before with clones using the CH340G.

rhfinch, 2017-07-26:
Thanks for replying. 

There is no serial to USB. The laptop com port is connected to the arduino  Rx Tx pins through a level converter. Are you saying that BCNC could be reading corrupted data from the arduino? What data does bcnc read while sending a file? I see no sign of it in the terminal window.

electrokean, 2017-07-26:
Yes, it appears that bCNC received corrupted data from the Arduino.  That section of Sender.py code is dealing with receiving status reports and parsing them (your line number doesn't exactly match the current code, but it is in the same area): https://github.com/vlachoudis/bCNC/blob/master/Sender.py#L1007

The status response from grbl in your exception output above is "-121.04-0.500" - which appears corrupted, as it is missing at minimum a comma between (I believe) the Y & Z coordinates.  Thus bCNC had an error parsing the invalid number for the MPos Y value.

Maybe your laptop is struggling to keep up with the serial data, or the level converter could be missing the occasional bit and causing data corruption.  Or possibly just dropped bits due to electrical noise.

rhfinch, 2017-07-26:
OK. Thanks for that. Now I know ehat to look forr.

ghlambert, 2017-07-26:
I have the same problem when my plasma cutter. My plasma cutter send a high voltage to start the plasma with the metal plate. At this moment, the message can be corrupted. 
You can add a few lines of code (try : ... except ValueError: ...continue) to reject corrupted messages. See below : 
if word[0] == "MPos":
    try :
        CNC.vars["mx"] = float(word[1])
        CNC.vars["my"] = float(word[2])
        CNC.vars["mz"] = float(word[3])
    except ValueError:
        print "garbage receive MPos:" , line
        continue
	CNC.vars["wx"] = round(CNC.vars["mx"]-CNC.vars["wcox"], CNC.digits)
	CNC.vars["wy"] = round(CNC.vars["my"]-CNC.vars["wcoy"], CNC.digits)
	CNC.vars["wz"] = round(CNC.vars["mz"]-CNC.vars["wcoz"], CNC.digits)
	self._posUpdate = True

elif word[0] == "F":
    try :                                                                
        CNC.vars["curfeed"] = float(word[1])
    except ValueError:
        print "garbage receive F:" , line
        continue

elif word[0] == "FS":
    try :                                                                
        CNC.vars["curfeed"]    = float(word[1])
        CNC.vars["curspindle"] = float(word[2])
    except ValueError:
        print "garbage receive FS:" , line
        continue
	
elif word[0] == "Bf":
    try :                                                                
        CNC.vars["planner"] = int(word[1])
        CNC.vars["rxbytes"] = int(word[2])
    except ValueError:
        print "garbage receive Bf:" , line
        continue
	
elif word[0] == "Ov":
    try :                                                                
        CNC.vars["OvFeed"]    = int(word[1])
        CNC.vars["OvRapid"]   = int(word[2])
        CNC.vars["OvSpindle"] = int(word[2])
    except ValueError:
        print "garbage receive Ov:" , line
        continue
	
elif word[0] == "WCO":
    try :                                                                
        CNC.vars["wcox"] = float(word[1])
        CNC.vars["wcoy"] = float(word[2])
        CNC.vars["wcoz"] = float(word[3])
    except ValueError:
        print "garbage receive WCO:" , line
        continue

rhfinch, 2017-07-27:
Thanks ghlambert! That's great. I'm up and running now.

vlachoudis, 2017-08-02:
Thanks @ghlambert I've added it in the code

closed 2017-08-02 by vlachoudis

rhfinch, 2017-08-02:
@vlachoudis I have changed excepts to include IndexError to catch all
errors.

vlachoudis, 2017-08-03:
Good point!

khomouda, 2019-02-26:
I have an OX CNC that is working fine through bCNC but got this garbage error twice, today was one of them, I continued from where it stopped, but the problem is that the motor continues working forever when it happens when the CNC has stopped. I can see that the above code has not stopped it, if it is already inserted. also is there a way to stop the spindle when that happens?

Harvie, 2019-02-26:
> is there a way to stop the spindle when that happens?

Have you tried clicking on "hard reset"?

khomouda, 2019-02-26:
I meant that the spindle stops if this happens spontaneously if nobody is in front of the CNC, and the machine holds it's position to use it continue from where it stopped

Harvie, 2019-02-26:
I've looked into the code at the "garbage" error, i can reccomend you following:

1.) Make sure you run latest GRBL v1.1 (upgrade if needed)
2.) Make sure to run latest git master bCNC
3.) Make sure to use correct `$10` setting of GRBL https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#10---status-report-mask
4.) Make sure that there is no electromagnetic noise in your serial line