vlachoudis/bCNC #624: not connectet

Baeckchen, 2017-07-28:
Hello,
first of all, bcnc is a gread programm, thanks for all the work.

Today i update / upgrade my rasperry jessi and now bcnc prompt "not connectet" and the MPos and WPos displays are empty. But I can jog, homing etc. from bcnc and when I open the terminal Status and MPos are reportet.
Bcnc Version is 0.9.9
What can I do
Gruß aus Düsseldorf
Andreas

[image]

Effer, 2017-07-28:
Hi @Baeckchen,

do you set the correct firmware version to match your controller?

[image]

Baeckchen, 2017-07-28:
Hi @Effer,
I've nothing changed in bCNC and off course I've tested both GRBL versions.
At the End I've set up a new Jessy on another RPi and there are the same erros. I've never watched at the terminal screen are there only the MPos data and status or additonally the WPos data ?
Please check this for me.

vlachoudis, 2017-08-02:
Can you show a screenshot of the terminal

W-E, 2017-08-03:
There is a known issue with a linux kernel version and serial port. If you look in the terminal from which you started bCNC, you may see a message saying something like "device reports readiness to read but returned no data" or the like. This is related to the serial port driver in the Linux kernel. Googeling the exact message leads to solutions. Like https://askubuntu.com/questions/680473/ubuntu-14-04-usb-serial-port-device-reports-readiness-to-read-but-returned-n 
I had the same issue on an laptop running an not up-to-date Ubuntu. After a dist-upgrade ( which included a kernel upgrade) it works.
This may also be an issue on Raspian...

Baeckchen, 2017-08-11:
Hello,
back from holidays I`ve made a screenshot from the terminal after a homing command from bcnc.
[image]

closed 2017-08-11 by Baeckchen

Baeckchen, 2017-08-11:
and here the release info 

pi@raspberrypi:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.9.35-v7+ #1014 SMP Fri Jun 30 14:47:43 BST 2017 armv7l GNU/Linux

vlachoudis, 2017-08-11:
It seems you have changed the report setting in grbl v0. and you don't show the WPos that it is needed by bCNC.

Baeckchen, 2017-08-11:
I thing so, but I´ve made no changes in the grbl controller.
The only thing I´ve done is update and upgade the os and perhaps I´ve made changes to the grbl controller by pressing the <controller> button ?

[image]

vlachoudis, 2017-08-11:
Be careful the setup here is for Grbl V1 and not of V0.9.
Anyhow I would strongly recommend you upgrade your controller to V1.1 it has a lot of extra functionality that you could benefit of.

Baeckchen, 2017-08-11:
I`m such a idiot, I´ve changd the settings <$10 Status report option> to 1 back to 3 everything worked fine.

Never change a running system :-) so I do not dare to upgrade my controller to V1.1

Thankes for all the help

Grüße aus Düsseldorf
Andreas

vlachoudis, 2017-08-11:
On the contrary this upgrade you should do :)

github051, 2018-12-23:
Hi Andreas, 
I'm also came across same issue. Then i saw u r above comment and i corrected $10=3
Now its working fine.
Thank you.