vlachoudis/bCNC #625: G-Code file size

dmon1234, 2017-07-28:
Hi

I have been using bCNC for a bit now and for the most part it is great.  I am having an issue with gcode generated by Cut2D from Vectric.  The gcode produced for the GRBL-mm post processor is HUGE.  For instance, to run a job which engraves a clock face with a 1/8 round nose bit (1 pass): Using f-Engrave the generated file is 202 KB.  It loads and runs fine in bCNC.  The vectric generated file is almost 3MB in size.  bCNC takes a while loading it.  When the preview finally comes up but it is all screwed up.  Is there a limit in the number of gcode lines?  Any thoughts would be much appreciated.

With many thanks,
D

Effer, 2017-07-28:
Hi @dmon1234 ,
this is an issue that periodically rise up. 
bCNC use python tkinter to render the GUI and the canvas with all the path. This is not so optimized for a huge amount of lines, for this reason there's a small combobox  on top right corner of the canvas where you can set a maximum time (in seconds) to render the the gcode program.
If the rendere take longer than this, the draw routine is interrupted and a small warning message is show in the bottom. You can rise this value as much as you need, but the program will slow down if your cpu can't handle.
Another option is to disable the drawing of the canvas using the small button on the middle.

[image]

If this fix the issue please feel free to close the issue.

dmon1234, 2017-07-31:
Hi Effer,

Thanks for the reply.  I guess that my laptop is not fast enough to render (it is quite old).  I have disabled drawing and is all good.  Thanks for getting back.

Regards

closed 2017-07-31 by dmon1234