vlachoudis/bCNC #626: Serious issue!

ArcAIN6, 2017-08-01:
So i decided to give bCNC a try, and it's been a nightmare. 

The issue is, no matter what i do, as soon as i start to run a gcode file, bCNC runs off and crashes into the top-right corner of the machine at full speed. 

I've tried 4 different NC files, all that worked perfectly fine in zapmaker's g-code sender. 

I've also noticed, that bCNC has a red box in the visualizer that is nowhere near where the model is for the part. 

I'm really not understanding what is going on here... If i position the tool at the stock, then zero the machine, why is it flying off to crash in the top-right corner of the machine at full tilt?

Steps to reproduce:
1) Load NC file
2) Set stock on machine, and set tool tip to top of stock center. 
3) zero all axis
4) set feed rate to 250mm/min
5) Set spindle speed to 6500rpm, turn spindle on
6) Push "play" button
7) Watch in horror as your spindle shoots away from the stock, and crashes into the top-right corner of the machine (happened once, i've since hit the emergency stop button to save from breaking the machine)

So... What exactly is going on here? Is it ignoring the position i set? Is it using the machine position rather than the work position as it should? 
What is the red box? Is that supposed to be the work position? If so, why does it not move to the location of the model when i zero the axis out? 

I'm seriously at a loss here..

vlachoudis, 2017-08-02:
bCNC should send the gcode as it is to your controller, unless if you asked for advanced features like autoleveling or tool change. Could you post your gcode to check.

electrokean, 2017-08-02:
@ArcAIN6 how are you producing the g-code?  Some CAM tools will start with a G28 which will move to machine home, which you probably don't want.  For example in Fusion 360 you may need to edit the NC post to disable G28.
Regarding your other issue, if bCNC isn't showing the full model, it is probably just hitting the timeout for drawing - see https://github.com/vlachoudis/bCNC/issues/625 for details on how to adjust the timeout
Read/search through some of the other closed issues here - you will find solutions to these and most other common newbie problems. :)

ArcAIN6, 2017-08-03:
Using solidworks w/ hsmworks to produce the g-code. (file attached)
[Brace_Pre-Drill.zip](https://github.com/vlachoudis/bCNC/files/1197566/Brace_Pre-Drill.zip)

I've checked and G28 is not being added. 
That being said. homing the machine shouldn't send it into the opposite direction should it, as i always home the machine before jogging to position and zeroing the WCS. 

As for the second part. The full model is being shown, however, like i said, the red box is not centered on the model. If the origin is set to the center top of the stock, should not also the work area be centered to the model?

vlachoudis, 2017-08-15:
With no further info, I am closing the issue. Please reopen if needed

closed 2017-08-15 by vlachoudis