vlachoudis/bCNC #627: bCNC used with Grbl laser mode - use m4 instead of m3

W-E, 2017-08-02:
Grbl now has a very nice feature in the laser mode:
If you turn on the laser with "m3", it runs at the power specified by the Sxxx command
BUT if you turn on the laser by issuing an "m4", the laser power is adapted to the movement speed. This is great as othwerwise in and near corners, where the laser head moves slower, the laser tends to over-burn.

Therefore a feature request: Make a selection checkbox for laser mode "constant power" or "adaptive power" which then generates M3 or M4.
Ideally it even re-generates when you change this checkbox while an existing G-code file is already loaded.

I now hardcoded the "m4" in line 970 in "CNC.py" but an option checkbox would be much nicer, of course.
Sadly I can only write some C and Perl but no Python (yet... )

vlachoudis, 2017-08-03:
Done. I've add it in the Config as laser adaptive power

closed 2017-08-03 by vlachoudis

W-E, 2017-08-03:
vlachoudis,

Thanks a lot. So, it now is hard coded to M4 ? That's OK for me, but likely others would like to see M3..

Anyway, thanks a lot. THAT was quick !

W-E, 2017-08-03:
> Hi, why was my previous request not looked upon ? It is somehow on the
> same line of this.
>
> Here is the data: "#552
> <https://github.com/vlachoudis/bCNC/issues/552> Open Giochia opened
> this issue on 29 Mar · 6 comments" and the description was " it would
> be useful to have no Z movement when the bCNC is in LASER mode. This
> would save time in having to edit long lines of code when passing from
> mechanical Spindle to Laser use, with the identical Gcode item.
> Alternatively, if movement is not easily removable, it could be
> sufficient to turn off Laser during "Z" axis movements, so that the
> Laser beam does not insist in the same spot, creating a hole, while Z
> is moving."

No idea. I've seen this. Besides the fact that i BELIEVE most laser 
people have laser cutters that simply not have a z-axis at all ( like 
mine) and therefore they have no issue; It still is an open source 
project where Vlachoudis puts his spare time into something he then 
donates for free to all of us. So, there is NO valid reason to expect 
ANYTHING from him.

I was very surprised that he did this so fast. I would have expected 
some weeks delay or no reaction all. Not because I hav a bad opinion 
about Vlachoudis but because I also manage some ( non-public ) software 
in my spare time and know what that means.

So, as a little thank you to Vlachoudis I will work on the grman 
localization of bCNC. In my free time. So, no timelines given...

With best regards,
     Wolfgang

vlachoudis, 2017-08-03:
@Giochia this was something very easy to be implemented, what what you've asked in #552 I cannot say that I fully understood. Moreover, what I've thought I understood it is more complicated, than the present request.

Giochia, 2017-08-03:
Hello,

it is in fact very simple indeed. It just require a flag to inhibit Z 
axis movement. This can be flagged on to avoid any Z axis motor action 
or flagged off to have normal operation. I believe it must be easy to 
implement.

Ciao, Giorgio

---
Questa e-mail è stata controllata per individuare virus con Avast antivirus.
https://www.avast.com/antivirus

vlachoudis, 2017-08-04:
Ok I see, you want an option to remove as a filter all Z movements. Is not complicated but not as straight forward like the m3/m4 implementation. It will require to go through the gcode and strip out any Z motion, and clean empty lines.

W-E, 2017-08-04:
Dear Vasilis,

I believe the request is not necessarily to clear all Z movements. The 
goal is just to not have any. So, maybe it would be possible to simply 
set them all to zero ?
Or to not generate them upfront ?
Or to somehow not send them in the sender routines, albeit they are in 
the G-code ?

Wolfgang

vlachoudis, 2017-08-04:
@W-E when you set the lasercutter mode, bCNC should not generate any Z movement for any generated gcode by bCNC. However it wont alter any existing/loaded g-code that contains Z-movements.

I am trying also to keep to minimum the changes it does to an existing/loaded gcode, for the moment is only on the autoleveling, tool changes and canned cycles to support extra features that are not existing in grbl.
So as I see it the best way is to create a filter, in the "Editor" to remove all Z movements from a loaded gcode.

Giochia, 2017-08-04:
CNC machines come in many shapes and sizes, in particular, some, like 
mine, have twin head, laser and cutter. Having to modify file, by 
editing manually, is a long an sometimes difficult task. And it means 
that we must have two files for a single job, one with Z movement and 
one without for laser mode. I feel that a flag to ignore any Z data 
could be a very easy alternative thus making bCNC even more complete. 
But I do understand that it might be more complicated than it seems.

Giorgio

---
Questa e-mail è stata controllata per individuare virus con Avast antivirus.
https://www.avast.com/antivirus

fbonin5491, 2023-08-21:
when you use "replace all" in a text editor ,  its nor so long.