vlachoudis/bCNC #628: bCNC always resets controller settings to defaults - but only under debian linux

W-E, 2017-08-02:
Whenever I change controller settings ( grbl settings) in "Tools" - "Controller", they are reset to some defaults as soon as I switch from "Controller" to any other function/section and then back.
Also, I verified by testing that - at least for the "Lasermode" Checkbox ( should set $32=1) - it has no effect.
So, you load a g-code file, go to "Settings" - "Controller" and check the "Laser mode enable" checkbox. Then you go to "Control" tab and start the program. The LaserMode is not used !
If you issue "$32=1" in the command line before running the program, it works.
It does not help to save "the file" at any time. The settings in "Tools" - "Controller" are always reset.
Even if I change these settings in ~/.bCNC with a text editor, I can see them after starting bCNC but after switching from "Tools" - "Controller" to anything else and then back, the values are reset. And then overwritten in ~/.bCNC . It looks like it is re-writte on program START ??? Where does it come from ?
So, it is impossible to set any Controller value via bCNC.
The strange thing is: This does not happen on Windows ( Win7 ) but on Debian9. I have bCNC on test on both platforms.
Newest bCNC Version, downloaded 2 days ago...

vlachoudis, 2017-08-03:
You have to send the configuration to the controller, by clicking the button "Controller" on top of the list

W-E, 2017-08-03:
Well, the "controller" thing on top of the options list is not clickable here. 
To avoid furter confusion, this time I did connect a controller running grbl to test this.
If I clear the terminal, then go to "Tools" - "Controller" , change $130 from 560 to 200, click the "Controller" button ( both, the non-clickable above the list and the one in the tab) and go back to the Terminal, this is what I get:
$$
$0=10
$1=25
$2=0
$3=0
$4=0
$5=0
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=0
$22=0
$23=0
$24=25.000
$25=500.000
$26=250
$27=1.000
$30=1000
$31=0
$32=1
$100=80.000
$101=80.000
$102=80.000
$110=2000.000
$111=2000.000
$112=500.000
$120=30.000
$121=30.000
$122=10.000
$130=560.000
$131=500.000
$132=200.000
ok
$G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
$$
$0=10
$1=25
$2=0
$3=0
$4=0
$5=0
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=0
$22=0
$23=0
$24=25.000
$25=500.000
$26=250
$27=1.000
$30=1000
$31=0
$32=1
$100=80.000
$101=80.000
$102=80.000
$110=2000.000
$111=2000.000
$112=500.000
$120=30.000
$121=30.000
$122=10.000
$130=560.000
$131=500.000
$132=200.000
ok
$G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok

Coloring is lost here. Only the $$ and $G are colored, so bCNC does only send $$ and $G but never $aaa=bbb
The "$$" is issued two times; once when I come from terminal to "Tools" - " Controler" and once when I click "Tools" - "Controller" again.
Anyhow, for now I set the values manually in the command line. As grbl remembers them, the problem is solved for me for now.
I still would consider this a bug, because it is not possible to change grbl parameters from within the bCNC GUI by modifying the controller options list.

Besides that, great program, thanks a lot. I dicover new great features all the time. Like saving existing .ngc files as .dxf. Works ! Great !

vlachoudis, 2017-08-04:
For some reason the button was disabled. Maybe when I did some changes in the Tools page. Now it is re-enabled so you can send to the controller the modified setup.

vlachoudis, 2017-08-04:
OK, found it was with the commit when I've updated the error messages list of grbl

closed 2017-08-04 by vlachoudis

W-E, 2017-08-04:
Hi Vlachoudis, thank you a lot for this super quick responses to all the issues I opened.
Sorry for that. 
Great work !