vlachoudis/bCNC #630: Autoleveling: can we get faster XY movements?

doegox, 2017-08-03:
Hi

When autoprobing with my fragile 0.1mm drill I'm happy to set Z probe feed at F10.
But X & Y run at the same super slow speed.
Am I missing something in the configuration or is it normal?
If it's normal, could we have the choice to select another feed for the X & Y movements?

lalo-uy, 2017-08-03:
X Y moves should be done with G0 so they go at max speed.

doegox, 2017-08-03:
Hmm ok I see the source of the problem:
```
G0X1F300      => moves at 300
G38.2Z-5F10  => probes at 10
G0Z5               => moves at 10
```

My CNC runs Smoothieware and visibly it assumes a F in G38.2 changes the current (G0) feed.
I don't know if smoothieware is wrong on that or not, I didn't find a proper spec. Linuxcnc speaks about "at current feed rate" for G38.2.
Anyways it would be more robust if bCNC appended systematically some F to the G0 commands during the autoprobing. It would be transparent for users of other firmwares.
What do you think?

vlachoudis, 2017-08-04:
There was a option for that. You activate it from the command line
```FEED ON```
then it appends the current feed rate on every line.
It was used in the past for having a faster response with feed overrides in grbl V0.9

doegox, 2017-08-05:
Hi
Hmm I tried, I see indeed the message "Feed appending turned on".
But during autolevelling this is not happening:
```
G0Z3
ok
G0X1.17Y3.17
ok
G0Z0.5
ok
G0X1.17Y3.17
ok
G38.2Z-5F10
[PRB:1.170,3.170,0.153:1]
ok
G0Z0.5
ok
G0X9.09Y3.17
ok
G38.2Z-5F10
[PRB:9.090,3.170,-0.746:1]
ok
G0Z0.5
ok
```
etc

The patch I'm now using locally to solve this problem is:
```diff
@@ -280,7 +280,7 @@ class Probe:
 		for j in range(self.yn):
 			y = self.ymin + self._ystep*j
 			for i in range(self.xn):
-				lines.append("G0Z%.4f"%(self.zmax))
+				lines.append("G0Z%.4fF%g"%(self.zmax, CNC.vars["fastprbfeed"]))
 				lines.append("G0X%.4fY%.4f"%(x,y))
 				lines.append("%wait")	# added for smoothie
 				lines.append("%sZ%.4fF%g"%(CNC.vars["prbcmd"], self.zmin, CNC.vars["prbfeed"]))
@@ -288,7 +288,7 @@ class Probe:
 				x += xstep
 			x -= xstep
 			xstep = -xstep
-		lines.append("G0Z%.4f"%(self.zmax))
+		lines.append("G0Z%.4fF%g"%(self.zmax,CNC.vars["fastprbfeed"]))
 		lines.append("G0X%.4fY%.4f"%(self.xmin,self.ymin))
 		return lines
 ```
It's obviously hacky and is reusing the fastprbfeed setting as a G0 feed.

vlachoudis, 2017-08-09:
G0 are using the rapid feed speed. Except on smoothieware that G0 has a different feed than G1/2/3.
In this case you can set the G0 globally before launching the command.

vlachoudis, 2017-08-09:
I was too fast, it seems that the G38.x alters the G0 feed. The "fastprbfeed" is not a rapid movement feed, but rather a fast probe feed. I don't have anywhere in the settings a G0 feed.

doegox, 2017-08-09:
yes "fastprbfeed" was a workaround because I didn't find quickly a setting with current G0 feed

vlachoudis, 2017-08-09:
The  CNC.feedmax_x can be used instead

doegox, 2017-08-09:
ha ok thanks!