vlachoudis/bCNC #631: [Feature request] Better support visualisation of large gcode files.

hb020, 2017-08-04:
When using anything larger than a thousand G code statements, bCNC becomes extremely sluggish (on my MBP, no this is not on RPi). When switching off "display of paths (G1,G2,G3)", suddenly it becomes responsive again. Also there are no longer sudden long lags while milling. As most of my mills are way above 1000 statements, I have to switch that off all the time and I am missing a lot of functionality.

This might be solved for example by making CNCCanvas.drawPaths() run in separate thread, although I do not quite know if that would break a lot of things.

vlachoudis, 2017-08-04:
Its already the case but in reverse. Since Tkinter is not compiled with multi-thread support in many OS is not possible to have any drawing operation in a background thread. Thus all drawing operations are performed on the master thread, and all I/O operations are performed in a background thread. Up to now I have never experienced any lags during milling, even on huge G-code files.
On which OS do you run?

hb020, 2017-08-05:
I am running bCNC on latest OSX, the OS standard python. 
So I tested on other environments:

- brew python 2 (with multithreaded tk, using -devel, will crash otherwise): slightly longer.
- Ubuntu 16 (in a VM): somewhat faster (about 2/3 of the time needed on OSX).

Still, the time needed for drawing is long. A 'small" 900kb gcode file takes 6 seconds to redraw on OSX, during which time the UI is non responsive. 
The lags during milling have happened to me several times (and when milling plastics that is a sure fire way to thrash the work). Will test from ubuntu.  Also, with drawing disabled, I have not experienced those lags.
I know I can set the timeout in the GUI, but then I get incomplete displays, which is misleading.
I understand that with single threaded tk, things are difficult. There are several options though.

vlachoudis, 2017-08-11:
Just to clarify, the milling runs on its own thread, so it is not affected at all by the drawing operations.
There were some lags during some display operations (#632).
I've "fixed" it by adding forced periodic updates to Tk.
Try the new commit and let me know your comments.

hb020, 2017-08-13:
well, that didn't go well. Now when triggered from bCanvas.draw(), it is making the app go into 100% CPU (on the one thread), never painting the entire image, never returning from self.update(), and blocking the UI.

vlachoudis, 2017-08-14:
Can you send me your example of large g-code, so I can check directly on that.

hb020, 2017-08-14:
my example: (rename to .ngc)
[Counter-F.Cu.txt](https://github.com/vlachoudis/bCNC/files/1222216/Counter-F.Cu.txt)

vlachoudis, 2017-08-15:
Unfortunately I cannot test on OSX. On Ubuntu that I have it works ok.

vlachoudis, 2017-08-15:
Can you check what #613 did maybe it helps?

rfsouzax, 2017-09-29:
Hello, I'm also experiencing the same problem when I use very large codes, I wonder if it would be possible to increase the loading time to render my file, here is the message: "RENDERING TAKES TOO LONG ... INTERRUPTED ...".
Any idea how we can fix this?
Thank you!
[image]

lalo-uy, 2017-09-29:
There is a parameter for that set to 5 sec by default.
Enlarge it