vlachoudis/bCNC #632: Send gcode will pause for a few seconds.

viewsat, 2017-08-05:
When I use Raspberry pi 2B to send a larger number of gcode, bcnc will have a pause after a few seconds to continue to send, what is the reason for this?
I use the gcode to test >> [LON.ZIP](https://github.com/vlachoudis/bCNC/files/1202030/LON.ZIP)

vlachoudis, 2017-08-11:
I am trying to investigate the freezing

vlachoudis, 2017-08-11:
I've found the culprit. After submitting the run pushing the gcode to the separate thread the programs tries to restore all paths. This loop just changing the color of the line is taking a lot of time! Now I am forcing a periodic update of the display so you don't see any freeze.

viewsat, 2017-08-12:
@vlachoudis  I use the Raspberry pi 2B execution bcnc-master (2017-08-11) send LON.cnc this gcode, at 1m36s of time will stop sending a few seconds.

vlachoudis, 2017-08-12:
@viewsat are you sure that it stops sending or the display is freezing? Sending is performed on a separate thread, nothing to do with drawing. I don't have an RPi setup, but with my machine with a very old laptop 10y+ running ubuntu, it runs flawlessly to the end (aircutting) with the grbl buffer always full (>80%)

viewsat, 2017-08-13:
@vlachoudis Because in the Raspberry pi only bcnc work the best, hoping to be able to find this short pause to send the problem, making bcnc can be more perfect.

vlachoudis, 2017-08-13:
@viewsat can you check if it is a memory issue, maybe it starts swapping and freeze?

vlachoudis, 2017-08-15:
It seems the solution to work on linux, so I am closing the issue

closed 2017-08-15 by vlachoudis