vlachoudis/bCNC #633: Autolevel is not working anymore

truhlikfredy, 2017-08-06:
I used the git clone version today. And when I do autolevel scan, the Z moves to 0 and then refuses to move into the -Z. When I have manual control I can move into -Z easily. I setup the -10.0mm as min Z probing and 1mm as max.

vlachoudis, 2017-08-09:
Maybe a screen shot of the terminal could help or some additional information. I cannot understand why. Normally nothing has changed in the autolevel routine since some time now. I've made a test on my mockup and it worked ok.

truhlikfredy, 2017-08-10:
I tried it now and now it works, that is strange. I'm still building the hardware as well so the issues could be overlapping with something else. And I think I might have some hardware related issue where the device stops working at random times, Will try add ferrite cores and capacitors if it's from some noise related issue. Using it with the v0.9 and CyclonePcb. Probably let's close this and if I will be able to replicate it and document it better I will comment here again.

vlachoudis, 2017-08-11:
Ok, thanks for the report

closed 2017-08-11 by vlachoudis