vlachoudis/bCNC #634: Home shortcut shouldn't be on by default

truhlikfredy, 2017-08-06:
I know you meant it well, but in my case I use the home to edit numbers a lot and if the gantry is in such position which would damage the head/gears which is in my case very often then then HOME button is very scary.

Either remove these:

self.bind('<Home>',		self.home)
self.bind('<KP_Home>',	self.home)

Or allow them to be configured, but they shouldn't be on all the time by default.

vlachoudis, 2017-08-09:
Indeed I agree. its done. both KP_Home and KP_End are removed.

closed 2017-08-09 by vlachoudis

truhlikfredy, 2017-08-10:
I didn't had much problems with end, but it will be safer. What is actually the difference between Home and KP_HOME ? I meant to write <Home> in the first line and somehow only '' is there.

vlachoudis, 2017-08-11:
KP_xxxx = keypad

truhlikfredy, 2017-08-11:
Then probably the non-keypad should be removed as well?

vlachoudis, 2017-08-11:
correct, done