vlachoudis/bCNC #636: LASER : laser does not turn on when SPINDLE button in "Control" is used AND LaserMode activated

W-E, 2017-08-11:
This used to work before the changes ( which I requested) to support GRBL's adaptive laser power mode.
Now, if you want to turn the laser on with the "Spindle" button in the "control" section, e.g. to focus it, it does not work anymore.
You have to turn off the "LaserMode" in the GRBL (controller) settings pane to be able to turn the laser on while the X and Y axis do not move.
I will investigate further, but I believe to have read somthing about GRBL laser mode (as a new feature) not allowing to turn the laser on if neither X nor Y axis move. This is to avoid burns.

So, MAYBE bCNC needs to do this:
IF (bCNC_Lasermode) {
 IF (SpindleButtonOn)
    TurnOffGRBLLasermode
    IssueM3
  ELSE IF (SpindleButtonOff)
    Issue S0 and (whatever to turn off the spindle )
    TurnOnGRBLLaserMode
  ENDIF
ENDIF

kees1948, 2017-08-24:
Would be very helpful, yes

tklus, 2018-09-11:
Just started using bcnc with my laser and had a heck of a time figuring this out. 

I used to use grbl panel and with that program, an M3 command will turn the laser on, M5 off.

I found I had to not only turn laser mode off in grbl settings but also in the settings for bcnc.

I usually want to turn the laser on at low power to locate my 0,0 location. So I usually just send a M3S1 command to turn it on at the lowest setting.

This would be great if this issue could be fixed. Not sure how grbl panel does it but it works.

Thanks!
Tim