vlachoudis/bCNC #638: bCNC making strange circles

abhinndas, 2017-08-16:
Hi 
I have been using bCNC for a while, but today it made a lot of strange circles in the visualizer. I am using fusion 360 for generating g-code. 
[image]

vlachoudis, 2017-08-16:
There should be some numerical precision error in the g2/g3 treatment. Can you send me the gcode file?

abhinndas, 2017-08-16:
: (PGM, NAME="0125")
; T1  D=6 CR=0 - ZMIN=-27.5 - FLAT END MILL
: G90 G40 G94
G17
G71
M26
; 2D CONTOUR5
M9
M26
:T1 M6
M26
S32000 M3
H0
M9
G0 X-30.511 Y201.6
Z15
Z5.5
G1 Z-3.9 F1524
G18 G3 X-31.111 Z-4.5 I-31.111 K-3.9
G1 X-31.712
G17 G3 X-32.312 Y201 I-31.712 J201
G1 Y126
G2 X-33.19 Y123.879 I-35.312 J126
G1 X-83.19 Y73.879
G2 X-84.98 Y73.018 I-85.312 J76
G1 X-534.981 Y23.018
G2 X-535.643 I-535.312 J26
G1 X-985.643 Y73.018
G2 X-987.433 Y73.879 I-985.312 J76
G1 X-1037.431 Y123.879
G2 X-1038.31 Y126 I-1035.31 J126
G1 Y376
G2 X-1035.31 Y379 I-1035.31 J376
G1 X-984.812
G2 X-981.812 Y376 I-984.812 J376
G1 Y254
X-970.812
Y376
G2 X-967.812 Y379 I-967.812 J376
G1 X-543.812
G2 X-540.812 Y376 I-543.812 J376
G1 Y254
X-529.812
Y376
G2 X-526.812 Y379 I-526.812 J376
G1 X-102.812
G2 X-99.812 Y376 I-102.812 J376
G1 Y254
X-88.812
Y376
G2 X-85.812 Y379 I-85.812 J376
G1 X-35.312
G2 X-32.312 Y376 I-35.312 J376
G1 Y201
G3 X-31.712 Y200.4 I-31.712 J201
G1 X-31.111
G18 G2 X-30.511 Z-3.9 I-31.111 K-3.9
G0 Z5
Y201.6
G1 Z1 F1524
Z-8.4
G3 X-31.111 Z-9 I-31.111 K-8.4
G1 X-31.712
G17 G3 X-32.312 Y201 I-31.712 J201
G1 Y126
G2 X-33.19 Y123.879 I-35.312 J126
G1 X-83.19 Y73.879
G2 X-84.98 Y73.018 I-85.312 J76
G1 X-534.981 Y23.018
G2 X-535.643 I-535.312 J26
[... 227 lines cut ...]
G1 X-543.812
G2 X-540.812 Y376 I-543.812 J376
G1 Y254
X-529.812
Y376
G2 X-526.812 Y379 I-526.812 J376
G1 X-102.812
G2 X-99.812 Y376 I-102.812 J376
G1 Y254
X-88.812
Y376
G2 X-85.812 Y379 I-85.812 J376
G1 X-35.312
G2 X-32.312 Y376 I-35.312 J376
G1 Y201
G3 X-31.712 Y200.4 I-31.712 J201
G1 X-31.111
G18 G2 X-30.511 Z-26.9 I-31.111 K-26.9
G0 Z15
G17
M9
M26
G0 X0 Y0
M30
M2

abhinndas, 2017-08-16:
How can I rectify this error?

vlachoudis, 2017-08-16:
It seems that the gcode has a problem, not the viewer.
For testing I've reduce it up to the first G17 G3 line and I get the following plot
[image]
The G17 G3 asks for a CCW circle on X-Y plane starting from
X-31.712 Y201.6 Z-4.5
to
X-32.312 Y201 I-31.712 J201
Therefore the center should be at X-31.712-31.712  and Y201.6+201 on the top-left wrt to the starting position, therefore it needs to make a circle CCW of almost 358deg.
The mistake is on the I,J if it was negative as I31.712 J-201, then it comes out correct as circle with the center at bottom-right wrt the starting point.

I've test it also on other online viewers https://ncviewer.com/  and I am getting exactly the same plot like in bCNC.

abhinndas, 2017-08-16:
I tried it again, and now I am getting expected results. I may have missed out one of the steps while compiling the code. Thanks a lot for the prompt replies, you have been a lifesaver!

vlachoudis, 2017-08-16:
It seems like a bug in the cam program, it would be nice to reproduce and report to them.

closed 2017-08-16 by vlachoudis