vlachoudis/bCNC #640: No manual spindle control

adambrum, 2017-08-16:
Im still learning on bcnc but i cant seem to get manual spindle control working.

bcnc rev 0.9.0.1
grbl rev 1.1
running on Raspberry pi3.

I can control the spindle with g code but not with the manual control.

Thanks Adam

vlachoudis, 2017-08-17:
Pressing the Spindle button sends the relevant on/off mxxx command. However when on you need to drag the slide with the RPM. BTW you have the latest version of bCNC?

adambrum, 2017-08-17:
I don't get any commands when I press the spindle button at all.

vlachoudis, 2017-08-17:
Check if in the Tools->Config you have the "Laser Cutter" option enabled

adambrum, 2017-08-17:
I have tried that,

The spindle button does show as depressed when I click it but no commands sent

adambrum, 2017-08-19:
All my fault, works if machine is in idle mode.

Thanks for the help.

Just one other thing, when i overtravel after homing with soft limits enabled i get error 2 and the only way to clear it is to stop and restart communications ?

Any ideas

closed 2017-08-21 by vlachoudis