vlachoudis/bCNC #641: Losing serial connection

Guiz-git, 2017-08-16:
Hello ; 
First thanks for bcnc! Haven't used it much yet but looks great. I could do few parts without much problem related to bcnc.

Still, I got a problem. 
I am using a Rpi 2 B with raspbian and bcnc 0.9.10 to control a shapeoko3 (grbl 0,9).
I have no problem to connect, to jog, and so on but sometimes I just loose connection and all the push button to jog seems to stop working. After some tests I could find a way to repeat the bug very easily. I just have to use the Rpi for another task, for example opening the web browser while bcnc is connected. Then when I come back to bcnc the connection is lost and I have to close/open Bcnc and re-open to control the machine. This is as if the work load on the Rpi makes it loose the com.

I am getting a Rpi 3 to test and see if the incresed power can solve my problem but I'm not sure.

Do you have any idea ? Thanks,

vlachoudis, 2017-08-17:
My guess is that is due to the electrical noise produced by the RPi when is loaded. I have a similar problem with a RPi that when the CPU is loaded it makes a lot of noise on every GPIO pin and on the audio output. Try to shield better the serial connection as well provide a better ground to RPi.

Guiz-git, 2017-08-17:
OK..  I will try few things. 
But have already an USB isolator between the machine and the Pi.

Guiz-git, 2017-08-22:
OK, 

I have done the same test on the RPI3 and now everything is OK even if I load the PI

vlachoudis, 2017-08-22:
@Guiz-git if everything is Ok, we can close the issue?

Guiz-git, 2017-09-12:
of course !

closed 2017-09-12 by Guiz-git