vlachoudis/bCNC #642: Autoleveller

nealre, 2017-08-19:
When I get an error on autoleveller I can't reset the error? Tried several times to reset then tried to send a gcode and it would not accept it? (nothing happens) and i still can't reset. All i can do is restart the program.
Am i missing something?
Rick

TilTisback, 2017-08-20:
I can confirm that i had the same type of error once. Unfortunately i dont know exactly what caused the probe to fail.

vlachoudis, 2017-08-21:
DId you try the hard reset, on the drop down menu?

TilTisback, 2017-08-21:
I was able to reproduce the error:
Set up the autoleveling and connect the probe + and - wire with eachother. Then start the autoleveling. The probe will fail as expected but it cannot be reset with the alarm unlock and the soft reset. Hard reset from the dropdown however works fine. (i didnt know it was there).
I think it schould be resetable with the normal unlock/soft reset buttons.

scorpizz, 2018-10-17:
I know this is a very old thread but I have the same issue as mentioned by nealre and TilTisback.
When there's an error during autolevel (search range too small), the only way to clear the error is a hard reset.
Unfortunately this also destroys the xyz of MPOS and WPOS and therefor you need to do a new home-cyclus - it takes a looong time... ;-)
Is it possible to change this behavour, so the error state can be cleared by doing a soft reset instead og a hard reset ?
Running sw:
"bCNC"
__version__="0.9.14"
__date__="5 Feb 2018"

Harvie, 2018-10-29:
yes. i have this problem too and it's annoying. when something goes wrong when probing, it's hard to recover. any ideas why this happens?

Harvie, 2018-12-16:
related code:
`class Probe:` in `CNC.py`