vlachoudis/bCNC #644: Gcode Editor is mixing up instructions

TilTisback, 2017-08-20:
Hello,
i wanted to do some small operations on my CNC and typed some commands into the editor directly:

>g0 z1
>g0 x0 y0
>g0 x 5.2 y-2.4
>g1 z-1 F50
>g0 z1
>g0 x14.7 y-2.4
>g1 z-1 F50
>g0 z1
>g0 x0 y0

But when i ran the code (with a safe Z height fortunately) i saw it was totally wrong.
So i took a closer look on the gcode in the terminal and saw that is was sending the commands in the wrong order, despite the fact that it was in the correct order in the editor.

So i saved the code to a file and checked it. here is the result: 
>g0 z1
>g0 x0 y0
>g0 x 5.2 y-2.4
>g1 z-1 F50
>g1 z-1 F50
>g0 x0 y0
>g0 z1
>g0 z1
>g0 x14.7 y-2.4

So why is bCNC mixing up the code? It looks to me that it does some type or sorting on the code so same commands are following eachother? 
When i run Gcode produced by another program eg. loading a file directly this sort of error does not show up.

TilTisback, 2017-08-20:
Hi,
I did some more tests with the editor and found out that the error only occours when you use the shift+enter on an existing line.
When you write one line after each other the error wont show up.
[image]

Here i wrote some code with line numbers n000-n150  in increments of 10.
Then i added the Z movements by selecting the n120 line and shift+enter and naming them n121 122 etc. 
When i save the code it will be mixed up:

>(Block-name: block)
>(Block-expand: 1)
>(Block-enable: 1)
>n000 g0 z10
>n110 g0 x0y0
>n120 g0 x100
>n121 g0 z-10
>n131 g0 z-10
>n141 g0 z-10
>n151 g0 z-10
>n152 g0 z10
>n142 g0 z10
>n132 g0  z10
>n122 g0 z10
>n130 g0 y-100
>n140 g0 x0
>n150 g0 y0

Somehow all the added lines are below the n120 but not even in the right order. (i added them like n121 n122 then n131 n132 etc.)
Maybe this helps to find the bug in the editor.

vlachoudis, 2017-08-21:
There should be some bug in the editing. Do you remember who you edited, and/or if you used the undo/redo. Just to spot the faulty command.

TilTisback, 2017-08-21:
I didnt use undo/redo.
I did this:
click on n120, Ctrl+Enter, type the n121, Ctrl+Enter, type the n122
click on n130, etc...

vlachoudis, 2017-08-21:
Ok. I've spotted the error, and I've corrected it. During the first insert  was not updating the internal pointers that was syncing the listbox with the gcode.

vlachoudis, 2017-08-21:
You can try the latest master.

TilTisback, 2017-08-21:
Its working fine now.
Thanks for the quick fix!

closed 2017-08-22 by vlachoudis