vlachoudis/bCNC #646: importing DXF

dodolbysuround, 2017-08-23:
Hey there,
I have some troubles importing DXF from Inksape. Are they any tricks or any point to care about when I generate the DXF?
My problem tooday is that rectangles with rounded squares are considered as right angle by BCnc
Thank by advance

vlachoudis, 2017-08-24:
Can you send us the dxf you tried, maybe as well the svg from inkscape

dodolbysuround, 2017-08-25:
Here you are (-;

[logo.zip](https://github.com/vlachoudis/bCNC/files/1251165/logo.zip)

vlachoudis, 2017-08-25:
It is not bCNC that interprets wrongly is the dxf conversion the problem. If you open your dxf with any cad (e.g. librecad)
[image]

I would suggest to convert all objects to paths in inkscape Menu -> Path -> Object to path
```Shift-Control-C```
BTW How did you convert the svg to dxf?

dodolbysuround, 2017-08-25:
I just used "save as" / .DXF 
is there any better way to do?

vlachoudis, 2017-08-25:
Interesting I didn't know that :)
Anyway if you convert first to paths then it works.

closed 2017-08-25 by vlachoudis