vlachoudis/bCNC #647: Cannot connect to serial port

kirch93, 2017-08-26:
Hi Guys,

I have downloaded and installed python 2.7.13, pyserial 3.3 and BCNC, but cannot get anything to show up in the com port section. I also cant seem to get windows 7 to give me a com port number to type in manually either. I have a USB 2.0 type b plugged into my phoenix cnc controller which goes to the standard usb on the computer.

I am not sure if it is a problem with pyserial or not. Any help would be greatly appreciated.

vlachoudis, 2017-08-28:
You can try in the command line
C:> python
...
>>> import serial
and check the error message it will return

electrokean, 2017-08-28:
Check if the COM port appears under Ports in Windows Device Manager.  If it doesn't, or if it says it isn't working, then pyserial won't be able to use it.  You may need to install a driver for the CNC controller USB interface.

Harvie, 2018-10-29:
Please reopen if issue persists

closed 2018-10-29 by Harvie

JWood1w4, 2019-04-12:
I'm having the same issue. 
[image]

Here is what I've got as an error message.

I also don't have the drop down menu. The board shows up in the IDE and my device manager. Any ideas?

JWood1w4, 2019-04-12:
> 
> 
> You can try in the command line
> C:> python
> ...
> `>>> import serial`
> and check the error message it will return

I'm not getting anything back at all.

[image]

Harvie, 2019-04-12:
Have you installed bCNC using pip as reccomended in README and wiki?

JWood1w4, 2019-04-12:
I can get pyserial to work. I can get it to show com3. IDK what is wrong. 
[image]

Harvie, 2019-04-14:
Have you installed bCNC using pip as reccomended in README and wiki?
Can you provide output of `pip freeze`?

georgegohl888, 2020-06-26:
Ii have the same issue with windows 10