vlachoudis/bCNC #649: Wrong corner after job finished

raenji-sk, 2017-08-28:
Hi, I have a weird problem that after the job is finished the cnc goes to upper right corner of the machine.
I have set it up for lower left corner, homing is working correctly, the probing and job itself works great.
Maybe I missed something and it is not a bCNC problem. But anyway, does someone maybe have a solution for this?

electrokean, 2017-08-28:
Does your g-code end with a G28?  Some posts (like Fusion 360) default to including that.  It will move to machine zero rather than work position zero.

raenji-sk, 2017-08-28:
Thank you. 
I use Fusion 360 and the default GRBL postprocessor.
Looking at the G-code I see the G28 G91 Z0 and G28 X0 Y0 at the end. I will try to replace them with G0/G1 and give it another try when I have some time to play with it.

electrokean, 2017-08-28:
You can make a copy of the grbl postprocessor into your custom post folder and disable the G28 option in a text editor.  This works well for me.

raenji-sk, 2017-08-28:
Ok, I will try it. Thanks.

raenji-sk, 2017-09-12:
It did the trick, now it is OK.

closed 2017-09-12 by raenji-sk