vlachoudis/bCNC #650: Issue rotating G-Code

lakeroe, 2017-08-28:
Please try again rotating the attached G-Code, it's still not working.
This bug is related to https://github.com/vlachoudis/bCNC/issues/648

[Test.txt](https://github.com/vlachoudis/bCNC/files/1257211/Test.txt)

vlachoudis, 2017-08-28:
The problem is more deep. All editing (move, rotation, mirroring etc.) operations are based on 2D, while this one contains 3D circles in XZ and YZ planes, which clearly can be rotated at 90,180,270 degrees but not at arbitrary angles

lakeroe, 2017-08-29:
Okay thanks, but I don't really understand your answer.
You write

> which clearly can be rotated at 90,180,270 degrees but not at arbitrary angles

and that's exactly I want to do.
I just open the attached example and rotate it by 90 degress and that does not work.
Anyway thanks for this fantastic piece of software !

vlachoudis, 2017-08-29:
Circles in planes 18 (XZ),19 (YZ) can be rotated in right angles, but there is no equivalent command to have a circular path in a tilted plane. The only way would be to approximate it with a linear segments. Of course in your case it will work so I have to make some kind of exception to handle those cases.

lakeroe, 2017-08-29:
Okay thanks for the explanation, it's clear to me now.
For me it would be okay to only handle circular paths in the XZ and YZ planes and give some kind of information if rotating is not possible due to a tilted plane.