vlachoudis/bCNC #651: [NOT AN ISSUE] This is more of an information request

ArcAIN6, 2017-08-30:
Does the bCNC developers have any plans for implementing IP cameras in the future? I ask, because i have a nice wireless IP camera that i would like to use instead of USB (4' x 4' bed makes USB cable too long to be snaked through 2 sets of drag chains usefully)

vlachoudis, 2017-08-30:
It should be supported by opencv.org, the library that bCNC is using for accessing the camera. I never thought of it, to check or to implement the interface, since I don't have any iP camera.

ArcAIN6, 2017-08-31:
Unfortunately, i have yet to find any actual "settings" per say, for telling bcnc where to look for a camera.. as it's an IP camera, it's not physically attached / installed to the PC running bcnc, and i see no settings withing .bcnc to tell it where, or how to connect to the camera. 

It's not really that big of a deal, as i can connect to it through manycam, but that creates quite a bit of overhead, as manycam converts an rtsp stream on the fly from h.264 into a format it can processes and display within itself.

I've just been forced to split my gcode into multiple files, as complex toolchains causes bcnc to use a bit of resources, and with manycam going as well, they both pretty much gobble up all the system resources if i'm not careful.

Harvie, 2018-10-29:
I use cheap chinese "usb endoscope", you can get them with quite long cables and they are easy to insert into dragchain. Support for IP cameras might happen, but i guess it's very far from being a priority in few years, so i close this for now. Feel free to share pull request.

closed 2018-10-29 by Harvie