vlachoudis/bCNC #652: Conflict USB Camera with Serial Communication?

mallemann, 2017-08-31:
Hello
Great Job on bCNC! I use it on a Raspberry 3, everything works great so far. 
The only thing: I did install an USB Webcam. Everything worked, the Camera is on.

The Issue: 
If I turn the Camera on and the jog around, after some steps I can not longer jog. Sometimes after 3 times jog around, sometimes after 20 times. Also homing does not work then.

My Impression: could it be that the USB Cam has a conflict with the serial communication between that Raspberry and the CNC Hat?

Maybe you have an idea, would be great!

Regards

Effer, 2017-08-31:
Hi @mallemann ,
could be power consumption? Maybe you should try with a bigger power supply?

mallemann, 2017-08-31:
Hi @effer,
Don't think that's the issue. I use external Stepper drivers and power the Raspberry/CNC with a 5v/3A power supply. I use https://wiki.protoneer.co.nz/Raspberry_Pi_CNC but with external drivers instead of Pololu drivers. In this configuration, the Raspberry only needs the 5v power supply.

Here is my machine http://www.masdrive.ch/myprojects/?page_id=381, only difference to this doc is that I replaced the drv8825 drivers with external drivers.

The problem only occurs if the camera is on

mallemann, 2017-09-01:
In the meantime I tried what you (@ Effer) suggested. I connected the Raspberry on a seperate powerline with a 5V/12A power supply, completely seperated from the rest of the machines power suppplie/lines.
Still have the issue

Any ideas very welcome, thanks

Effer, 2017-09-01:
Does it happen the same with a different camera? I don't know what else suggest.

mallemann, 2017-09-01:
I will try to use a powered USB hub next and then another cam. Will let you know.

mallemann, 2017-09-26:
I was exploring here a little, the issue is still there:

- I used a powered USB hub
- I connected a brand new state of the Art USB Cam
- Raspberry is powered on separate Powerline and power supply (5V/3A), completely independent from the rest of the CNC (Motors, Spindle, etc.)

I am using the latest Raspberry with the Protoneer CNC hat (https://wiki.protoneer.co.nz/Raspberry_Pi_CNC)  and external motor drivers

**Issue is still there**: if i toggle to the camera view, the cam is on and works as expected. When I then start to jog around in bCNC, after a certain number of jog steps (between 2-20), jogging stops working. If i then close the serial port and reopen it again, it works again for some jogs and stops again.

It's is clearly a conflict on the serial port when the Cam is on.

What i found is: https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=182365. Looks like a similar issue but I dont know how this would help in my case using bCNC.

if somebody has an idea, greatly appreciated!

Thanks

mallemann, 2017-09-27:
Another Test: since the serial port configuration changed on the RPI 3: 
- I did connect on ttyS0
- I did connect on ttyAMA0
- I did disable Bluetooth and Wifi

nothing changed. Always same result: when the CAM is on, the serial bCNC is using to control the motors, got lost after 2-20 jogging steps. When the CAM is off (disconnected from USB), everything works fine.

no more ideas on my end. Somebody an idea? Thanks!

lalo-uy, 2017-09-27:
May be a power issue. Try the CAM on a powered USB Hub.

mallemann, 2017-09-28:
I did try also the powered USB hub, see my earlier post. Same issue.

vlachoudis, 2017-10-02:
I would also think of noise due to a power/grounding issue. I have one RPi controlling my home, and I've noticed that when the CPU was working it created a lot of noise, that was temporarily blocking the WiFi, bluetooth and my serial connection (through the GPIO) to an ATMEGA controlling all communication in the house. Noise which is audible if you connect a speaker with an amplifier on the 3.5mm jack. I was trying to debug it by listening to the noise on the speaker. I have not managed to eliminated it, but I've reduce it a bit by using better power supply and thicker cables for the ground of the RPi.

Unfortunately when the camera is enabled, bCNC is using heavily the CPU.

mallemann, 2017-10-03:
I did put the RPI on a separate Powerline and separate Power Supply, short and think wires. Maybe improved a little bit (not sure) but issue is not gone.

So my question:

is there anybody using the Camera in the same scenario successful? Just important to know if this works in other places before I invest to much time. My Concept:

I use bCNC: 
- on Raspberry Pi 3 
- with the protoneer CNC Hat V2.58 (embedded ATMEGA connected to the RPI3 GPIO)
- I use the preconfigured Image https://wiki.protoneer.co.nz/Raspberry_Pi_CNC_User_Interface_SD_Card_Image 
- that includes everything from bCNC to the serial port configuration
- I beliefe I provide proper ground to the RPI on separat power line / supply
- I use external drivers with their own power supply to drive the Nema 23 motors
- Tried 2 USB Cam's, connected directly or via Powered USB hub

Did not getting it to work. After 10-20 jog steps with Camera on, jogging does not longer work (serial connection is somehow corrupted, bCNC stills shows connected, but doesn't work anymore. Restart bCNC or close/open serial connection helps).
Without the Cam on, everything works great!

If somebody has a similar configuration that works, please let me know. Then I explore deeper on my end.

Thank you very much!!

mallemann, 2017-11-03:
I did not find a solution for this problem, still interested if somebody has an idea.

For now, I am using a separate RPI with the USB Camera, running nothing else than  a small python app to read the stream from the camera. This separat RPI has his own small screen just showing the video stream from the camera. bCNC runs on his own RPI completely independent from the RPI running the video stream. That works, no issues.