vlachoudis/bCNC #654: Bug in Version 0.9.10?

theblackraven, 2017-09-03:
Hi,
i tried today the newest version of bcnc  with grbl 1.1.
The status stays always connected, although the machine is running.
Pausing is possible, but you can't resume. 
With version 0.9.8 everything works fine. 

BTW:
I modded the Version 0.9.8 to use this with this jog wheel: http://www.ebay.com/itm/Digital-Hand-Wheel-HB04-3-Axis-MPG-Pendant-for-CNC-Pulse-Generator-Mach-3-Router/191310311229?_trkparms=aid%3D222007%26algo%3DSIM.MBE%26ao%3D1%26asc%3D46086%26meid%3Df70775569a2945b9acc910d2893afd60%26pid%3D100011%26rk%3D1%26rkt%3D5%26sd%3D151827232518&_trksid=p2047675.c100011.m1850

The code is very dirty, because i never used python before and jogging doesn't work perfectly.
I use the new jog commands from grbl 1.1.

Until now following things are possible:

- Jog
- Show wcs and mc coordinates on the display of the Jog Wheel
- Zero Axis
- Select Axis for Jogging
- Select three types for jogging:
     1. 0.002 mm per Step
     2. 0.05 mm per Step
     3. Automatically change feed and stepsize, depending on how fast you turn the wheel
- Set Min-Max feedrate
- Set increase and decrease factors, when feedrate is changed dynamically 

I have to patch following files to use the modded version:
 - sender
 - controller
 - cnc

And i added one file:
 - usb_hb04

Is there a way, that i could share the code?

vlachoudis, 2017-10-08:
You can make a pull requests from github, so I can check the code before commiting

hdo, 2018-06-28:
@theblackraven It's been a while but do you mind sharing your code for the pendant with us? (maybe gist?). I own a wireless XHC pendant and would like to integrate that into bCNC. Thank you.

theblackraven, 2018-09-30:
@hdo:
sorry for my late reply, but i read this request today for the first time.
you can download the modified code here: https://drive.google.com/open?id=1ngz2-tvLyq-DakHQgvVI2RgmZRfl1_5u

i don't know, if the code still works, because i skipped editing the code. 
The Code is very dirty, but the most import file should be this one:  usb_hb04,.py

the driver in the driver folder, only works with windows 7 and not with windows 10.
i tried it also with linux and it worked.

greetings theblackraven

hdo, 2018-10-01:
@theblackraven  Hey thanks for your reply. I noticed that bCNC can't handle jogging very well so i'm working on a dedicated jog controller for my GRBL board. BTW I can't access your code and sent you a request for accessing your drive.

Harvie, 2018-10-30:
Please try again with latest grbl and bCNC master

hdo, 2018-10-30:
Thanks Harvie. I checked the source of bCNC and i'm pretty sure that python cannot handle the critical timing for small steps jogging as described in GRBL1.1 very well. I'm currently working on a separate hardware jogging controller for the GRBL mega version.

Harvie, 2018-10-30:
If the joging wheel is based on sending bursts of individual steps to the motors, i guess it's not really ideal to try channel that through serial. But you might make some buffer, which will count the pulses of wheel for 0.1 or 0.05 second interval and then change feedrate accordingly to the burstrate of the steps.

hdo, 2018-10-31:
I'm doing exactly that. However i noticed that it's not smooth everytime. In some cases it misses the Jog cancel command but it's not related to bCNC though.

> If the joging wheel is based on sending bursts of individual steps to the motors, i guess it's not really ideal to try channel that through serial. But you might make some buffer, which will count the pulses of wheel for 0.1 or 0.05 second interval and then change feedrate accordingly to the burstrate of the steps. So the more steps you are getting, the faster feedrate you use. No need to send each step.

Harvie, 2018-10-31:
> In some cases it misses the Jog cancel command but it's not related to bCNC though.

Have you filled bug in grbl repository on that?

hdo, 2018-10-31:
There is already a ticket related to this :-)

> > In some cases it misses the Jog cancel command but it's not related to bCNC though.
> 
> Have you filled bug in grbl repository on that?

theblackraven, 2019-10-15:
Here is the last working bCNC with jogging support for the hb04 without a README, because i have no time for this. Feel free to improve and share :)

https://drive.google.com/file/d/1BfHiwTcXqmU6wGKfmIadTxpZpodU82Hf/view?usp=sharing

I think that i edited these Files:

- ControlPage
- Sender
- usb_hb04

Maybe you have to edit your udev rules for the USB Support or try it first as root :)

Harvie, 2019-10-15:
@theblackraven thanks for your involvement! Does your code fix/add something that is not yet present in git master? if so, would you mind creating pull request for us, so your fix will be beneficial for the rest of the community?

theblackraven, 2019-10-15:
@Harvie i didn't fix something in the code. I only added support for the hb04 jogging wheel in this version. But it's not really stable....