vlachoudis/bCNC #656: USB Connection fails

student1704, 2017-09-07:
The USB serial connection fails after a while.
It it isn't possible to reconnect except EEPROM contents is deleted by ARDUINO IDE

I work with GRBL 1.1 and W7 64 bit
Has anybody a hint?

chamnit, 2017-09-07:
USB disconnect has to do with electrical noise and it interrupting the communication between Grbl and the host computer. You fix it by shielding your USB cable from powered cables and other sources. And use a good quality cable with a ferrite core.

student1704, 2017-09-07:
It is.
But may be the switching PS is to closed to Arduino.

Giochia, 2017-09-08:
I had a similar problem, and after many tries, I found that it was the electrical motor powering the spindle that was introducing a lot of electrical noise. 
Try to disconnect the motor, and ensuring the bit does not hit any surface, and see if the problem persists. If it goes away, then you know what the cause is and you have to proceed with mounting a filter to the motor (capacitors).

student1704, 2017-09-11:
Thanks so far.
I will try that and keep you updated.
Just give me some time.

student1704, 2017-10-08:
It is very strange. The error doesn't occur during the job runs. It occurs while jogging.
With other GCODE sender it works independent from other failures they have.

Harvie, 2018-10-29:
is this still a problem?

vadimyar, 2024-06-26:
[SOLUTION (possibly...)](https://github.com/vlachoudis/bCNC/issues/1808#issuecomment-2100233973)