vlachoudis/bCNC #663: Smoothie new query format

wolfmanjm, 2017-09-27:
Smoothie now supports the new GRBL status format : 

```<Idle|MPos:0.0000,0.0000,0.0000,0.0000|WPos:0.0000,0.0000,0.0000|F:4000.0000>```

This will become the default in the future, right now it is selected in the config with...

```new_status_format true ```

It would be nice if bCNC could modify its handling of the status for this newer format.

-Thanks

wolfmanjm, 2017-09-27:
Please note it is not entirely the same as grbl 2 though, it has the old format with MPos and WPos, just uses the | to separate the fields as this is easier to parse than the old format.
It also allows 4 fields in the WPos for an A axis.

mattleaverton, 2017-11-11:
This change to bCNC breaks functionality on the latest stable Smoothie firmware acquired on 14-Oct. I tested with the latest bCNC today and it breaks when checking status on connect. Everything works well when using the commit previous to this one. If this is standard for Smoothie firmware in the future, it might be worth toggling this functionality in bCNC to support machines not on the bleeding edge (I cannot find a new_status_format in either project's configs - it only appears to be in the Smoothieware kernel).

wolfmanjm, 2017-11-11:
Just add new_status_format true to the config. it'll work with latest edge. This is the new format the old format is deprecated and will be removed from smoothie.
The wiki is not always up to date, you can add it if it bothers you it is a user supported wiki.

closed 2017-11-11 by wolfmanjm