vlachoudis/bCNC #666: RPI 3's main OS toolbar hides on bCNC start

xlnt-webdesign, 2017-10-02:
does anybody else has a problem with a raspberry 3 (current raspbian) and bCNC (current).
when I start bCNC the system toolbar starts flickering and finally disappears when bCNC is completely started/initialised/loaded.

is this a bug or a config issue?

thank for bCNC it is awesome
Daniel

joshcarz, 2017-10-09:
I have this same issue. The only way to get it back is to reboot. Anyone else?

vlachoudis, 2017-10-12:
What resolution your monitor has?

joshcarz, 2017-10-13:
My current resolution is 1360x768 although I have tried others. The main toolbar will flicker a few times while the program is loading then it totally disappears.

xlnt-webdesign, 2017-10-13:
My resolution are the "preferred monitor settings" (1920x1080@60Hz) on a common 24" business monitor with overscan disabled.
Changing these settings to other values doesn't help.

my current workaround is a terminal shortcut on the desktop and "sudo restart" to prevent rebooting by power off

joshcarz, 2017-10-13:
If I run bCNC.py directly the main toolbar flickers but doesnt disappear. It only is disappearing when I run the bCNC shell script. Not sure if this helps.

xlnt-webdesign, 2017-10-13:
I'm not 100% sure, but I think with Raspbian Jessie (version name) it worked well.
after I finished my CNC build, I updated everything including my raspberry Pi and raspbian to stretch (current version name). Since then, I have this issue every time I start bCNC

joshcarz, 2017-10-13:
run the bCNC.py directly? Does your main bar disappear then?

xlnt-webdesign, 2017-10-13:
@joshcarz, you are right, thanks a lot, It works for me too.

BUT if I resize the fully loaded bCNC window it flickers and disappears again.

joshcarz, 2017-10-13:
yup same here. I'm currently attempting to update python to see if that solves anything

joshcarz, 2017-10-13:
So python is already all up to date. sudo apt-get update and upgrade updated it already. 
Its funny because every once in awhile it will flicker and not disappear. Very strange.

xlnt-webdesign, 2017-10-13:
sudo apt-get update and upgrade is what I do regularly. This and copying bCNC from github is all I changed to the original raspbian image.

except from the default settings, like time, local and display settings

joshcarz, 2017-10-14:
So after some searching it sounds like this is an issue with the new version of pixel. I don't think this is a problem with bCNC. It really sucks though.

xlnt-webdesign, 2017-10-17:
for me that conclusion sounds to easy and might even be misleading.
A lot of UI apps running on the raspberry pi are written in python, it even is the preferred programming language for the raspberry.
I think that makes bCNC a first citizen for the pi

bCNC is a wonderful app, but the only one I have this crashing menu issue.

@joshcarz if that is a known problem with pixel that affects some apps, a workaround might exist already. Please share your researches to help vlachoudis close that bug

vlachoudis, 2017-10-20:
I am sorry but I cannot check, I don't have such a setup.

xlnt-webdesign, 2017-10-26:
@vlachoudis, In case you would like to maintain a bCNC compatibility with raspberry 3. I'm willing to donate a raspberry 3 Set to you.
I know that software can't run on every setup, but if raspberry is on your list, please tell me an easy way to donate.  paypal?

vlachoudis, 2017-10-26:
Thanks @xlnt-webdesign . It is not a question of RPi3. I have one, but headless and when I test there I connect via ssh exporting the graphics to my linux and everything works ok.

xlnt-webdesign, 2017-11-16:
I have updated my raspbian OS yesterday and the issue seems to be fixed by them. So it looks like @joshcarz was right, it was no issue with bCNC.
Can someone else please confirm this, just to be sure that none of my other changes had that effect.

thank you all

joshcarz, 2017-11-16:
did you update just with the update and upgrade commands? If so I'll try now.

joshcarz, 2017-11-16:
Yeah it looks like its fixed. Nice. Thanks for posting.

xlnt-webdesign, 2017-11-16:
perfect. I think this issue can be closed now.

vlachoudis, 2017-11-16:
Perfect. Thank you

closed 2017-11-16 by vlachoudis